peted-davis/WeatherFlow_PiConsole #76: Observations time out

cefoster0, 2022-12-13:
I've been observing for a month or so situations where the app stops receiving observations and puts up the (i) icon. Temperature and pressure do not change, but oddly it seems like wind direction seems to move.  Restarting the app will bring it back up, but the app is running on a dedicated PI that I don't have a keyboard on. Other systems (like weewx) don't seem to be dropping observations and there are no errors in Tempest app. The device its self is on the battery boost. 

 - OS: Raspberry Pi OS Debian 10.1
 - Hardware: PI4 4 GB
 - Console Version: latest

[image]

pjmpessers, 2022-12-13:
This happens to me also occasionally. After the new update i saw it ones. If you have a touchscreen the only thing you could do is push the menu and the exit. If you have WFPICONSOLE AUTOSTART-ENABLE it will restart and it receives the readings again without a problem.

Like i said. After the update i got it ones. And that is 6 days ago. So i don't know if it was a one time glitch or the problem still exist.

cefoster0, 2022-12-13:
I had the issue prior to the latest update, doesn't seem like the update changed anything. 

Unfortunately its on a non-touch screen TV and I have a xdecorations window over it so its rather hard to kick-over.

pjmpessers, 2022-12-13:
Can you reach it with a other computer with SSH? Then you can give it "sudo reboot"

peted-davis, 2022-12-15:
This is a known issue and is unfortunately related to a bug in the WeatherFlow API that means it sporadically stops sending data. I am adding a workaround into the next version

peted-davis, 2022-12-22:
This workaround has been added to v22.12.2

closed 2022-12-22 by peted-davis