peted-davis/WeatherFlow_PiConsole #77: crash when starting with 22.12.2

gee33, 2022-12-23:
since successful and complete update to 22.12.2 the console does crash while starting. 
The console appears for 2 seconds then crashes. 

System settings
  ---------------------------------
  Updating version number to: v22.12.2

 Traceback (most recent call last):
   File "main.py", line 521, in <module>
     wfpiconsole().run()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/app.py", line 950, in run
     runTouchApp()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 582, in runTouchApp
     EventLoop.mainloop()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 347, in mainloop
     self.idle()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 387, in idle
     Clock.tick()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/clock.py", line 733, in tick
     self.post_idle(ts, self.idle())
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/clock.py", line 776, in post_idle
     self._process_events()
   File "kivy/_clock.pyx", line 616, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 649, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 645, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 218, in kivy._clock.ClockEvent.tick
   File "/home/pi/wfpiconsole/lib/status.py", line 200, in get_device_status
     self.status_data['tempest_status']      = device_status
 UnboundLocalError: local variable 'device_status' referenced before assignment

after reboot same behaviour : 

Last login: Fri Dec 23 07:06:28 2022
pi@weatherraspi:~ $ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_22-12-23_18.txt
[INFO   ] [Kivy        ] v2.0.0
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.7/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Config      ] Verifying station details
[INFO   ] [Factory     ] 186 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: egl_rpi
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'OpenGL ES 2.0'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'VideoCore IV HW'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 0
[INFO   ] [GL          ] Shading version <b'OpenGL ES GLSL ES 1.00'>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <8>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [Websocket   ] 2022-12-23 07:08:07 - Opening connection
Unable to connect to X server
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [MTD         ] Read event from </dev/input/event0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [MTD         ] </dev/input/event0> range position X is 0 - 799
[INFO   ] [MTD         ] </dev/input/event0> range position Y is 0 - 479
[INFO   ] [HIDMotionEvent] using <raspberrypi-ts>
[INFO   ] [MTD         ] </dev/input/event0> range touch major is 0 - 0
[INFO   ] [HIDMotionEvent] <raspberrypi-ts> range ABS X position is 0 - 799
[INFO   ] [MTD         ] </dev/input/event0> range touch minor is 0 - 0
[INFO   ] [HIDMotionEvent] <raspberrypi-ts> range ABS Y position is 0 - 479
[INFO   ] [MTD         ] </dev/input/event0> range pressure is 0 - 255
[INFO   ] [HIDMotionEvent] <raspberrypi-ts> range position X is 0 - 799
[INFO   ] [MTD         ] </dev/input/event0> axes invertion: X is 0, Y is 0
[INFO   ] [HIDMotionEvent] <raspberrypi-ts> range position Y is 0 - 479
[INFO   ] [MTD         ] </dev/input/event0> rotation set to 0
[INFO   ] [Websocket   ] 2022-12-23 07:08:08 - Connection open
[INFO   ] [Base        ] Leaving application in progress...
 Traceback (most recent call last):
   File "main.py", line 521, in <module>
     wfpiconsole().run()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/app.py", line 950, in run
     runTouchApp()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 582, in runTouchApp
     EventLoop.mainloop()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 347, in mainloop
     self.idle()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 387, in idle
     Clock.tick()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/clock.py", line 733, in tick
     self.post_idle(ts, self.idle())
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/clock.py", line 776, in post_idle
     self._process_events()
   File "kivy/_clock.pyx", line 616, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 649, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 645, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 218, in kivy._clock.ClockEvent.tick
   File "/home/pi/wfpiconsole/lib/status.py", line 200, in get_device_status
     self.status_data['tempest_status']      = device_status
 UnboundLocalError: local variable 'device_status' referenced before assignment
pi@weatherraspi:~ $ 

thanks for looking into this and some hints.
G

peted-davis, 2022-12-23:
Thanks for the bug report. I will be releasing a new version soon that fixes this issue

mweilbacher63, 2022-12-23:
Thank you! I was running Pi Console on a Raspberry Pi Desktop OS Guest on
my Ubuntu Based Virtual Box VM for months. Then about a month ago, after
doing an update it stopped working. I decided to fall back to running Pi
Console on a Raspberry Pi3B+ Running Buster with a 7 inch HDMI screen
attached. I have run into multiple issues trying to get Pi Console to run
on the Pi3B+. I am running the standard install script per the
instructions. I have gotten errors about rustc not being installed.

peted-davis, 2022-12-23:
Try running this command before running the PiConsole install command:

`python3 -m pip install cryptography==37.0.4`

closed 2022-12-28 by peted-davis

mweilbacher63, 2022-12-28:
Running python3 -m pip install cryptography==37.0.4 resolved the
installation error. Should this be part of the install script, or
included in the installation instructions?

Thank you!!