peted-davis/WeatherFlow_PiConsole #78: WFPIConsole crashed and will not restart - KeyError: 'PanelSix' / ...  AttributeError: 'super' object has no attribute '__getattr__'

creas002, 2022-12-23:
WFPIConsole crashed and RPi reboot resulted in multiple errors as shown in attached log

Version = v22.12.1
Raspbian GNU/Linux 10 (buster)
RPi 4b - 4GB

[wfpiconsole_20221223.log](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10294701/wfpiconsole_20221223.log)

creas002, 2022-12-23:
I'm not sure why the app broken, perhaps with the latest library updates bumping version to 12.2.22? I ran **wfpiconsole update** and **sudo reboo**t and issue resolved.  Thx

peted-davis, 2022-12-23:
Great - you found the fix! There was a bug in the notification that shows in the console when an update is available. By releasing a new version to fix this bug, I also triggered this bug. Now you are running 22.12.2 it shouldn't happen again

closed 2022-12-23 by peted-davis

creas002, 2022-12-23:
Thank for the quick fix. Merry Christmas!