peted-davis/WeatherFlow_PiConsole #79: Can’t install

adamsfour, 2023-01-08:
**Describe the bug**
trying to install software. When I run from terminal I get the following

[image]

here is a snap shot of version 
[image]

Command I am running is curl -sSL https://peted-davis.github.io/w

peted-davis, 2023-01-08:
Not sure if you've made a typo in your post, but the full command you need to run is 

`curl -sSL https://peted-davis.github.io/wfpiconsole | bash`

adamsfour, 2023-01-08:
So that was the problem. Here is the line as it appears in instruction. Hence my confusion 
M
[image]

peted-davis, 2023-01-08:
Great! Glad we managed to find a quick fix. Uncle the GitHub README formats 'code' on a single line, so the grey box has a horizontal scroll bar if your screen is not wide enough to show the full command. It is easy to miss. Hope you enjoy the console!

closed 2023-01-08 by peted-davis

adamsfour, 2023-01-08:
Thanks one stupid question. Once installed can I reach the display from any computer on my network.