peted-davis/WeatherFlow_PiConsole #82: Install Script Crashes on Buster Pi 3B+

Death2Beavis, 2023-01-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Run install script "curl -sSL https://peted-davis.github.io/wfpiconsole | bash"
2. Click on OK to continue
3. Scroll down to (script runs, generates lots of debug)
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

Attached errors scraped from Pi terminal
[install errors.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10450526/install.errors.txt)

**Hardware/OS/Version (please complete the following information):**
 - OS: [e.g. Raspberry Pi OS Bullseye] Buster
 - Hardware: [e.g. Pi3] Pi3B+
 - Console Version: [e.g. 22.3.1] Unknown assume current

**Additional context**
This is a complete rebuild after console crashing (saw notes on bullseye, may be that issue). New SD card, new image of Buster, fully updated.

peted-davis, 2023-01-19:
Sorry, this is a known bug that I haven't had a chance to fix yet. 

Please run this command before running the install command

`python3 -m pip install cryptography==37.0.4`

Death2Beavis, 2023-01-19:
Thanks appreciate that

closed 2023-01-21 by peted-davis