peted-davis/WeatherFlow_PiConsole #86: Failed installation

adamsfour, 2023-02-10:
Attempting to install on pi3 see screenshot for the error I received. 
[image]
[image]
[image]
[image]

adamsfour, 2023-02-10:
I figured a way to see error log[ 
[ErrorLog.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10711361/ErrorLog.txt)

adamsfour, 2023-02-10:
Some logs 
[kivy_23-02-10_3.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10711666/kivy_23-02-10_3.txt)

adamsfour, 2023-02-10:
Sorry another data point. Did a clean install of OS, ran the extra script you mention in a closed issue (python - m pip ) then ran install. No change.

peted-davis, 2023-02-11:
I'm looking into this at the moment but I'm not having much luck I'm afraid. It looks like something has changed with Kivy and/or Raspberry Pi OS that is preventing Kivy from compiling correctly

adamsfour, 2023-02-11:
Okay. I did try install without taking OS upgrade but made no difference. Do we think it’s isolated to PI 3B or would I get same result on PI 4B. Not my first choice but could purchase a PI 4 if only way forward. Will stand by

adamsfour, 2023-02-11:
Would reverting to an older version of buster OS resolve the problem. I found earlier versions.

peted-davis, 2023-02-12:
I think this is only Pi3 related. You could definitely try an older version of Buster. It might help isolate the issue. I'll keep working on finding a fix for the Pi3, no need to get a Pi4 right now

adamsfour, 2023-02-12:
Tried 4-4-2022 buster build. Same error.

adamsfour, 2023-02-12:
Also tired 12-02-2021 build which gave me the same error. Thanks for your help

peted-davis, 2023-02-12:
Can you try running this command:

`python3 -m pip install https://kivy.org/downloads/2.1.0/Kivy-2.1.0.tar.gz`

If it works, it will take a while to compile

adamsfour, 2023-02-12:
Yes, I will report back once completed. I assume you want me to 1) run Python3 script and then follow by Curl script

peted-davis, 2023-02-12:
Yes, if the command I shared completes, then try the curl command again

adamsfour, 2023-02-12:
Sorry no luck. I have attached the screen dump and the error log. Hope this helps
[kivy_23-02-12_3.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10717213/kivy_23-02-12_3.txt)
[ErrorLog.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10717214/ErrorLog.txt)

peted-davis, 2023-02-12:
Ah, have you run the curl script at all on the current Raspberry Pi OS build you have your SD card? The curl script installs a bunch of dependencies that are required for the Python command I shared to run successfully. 

If you haven't, run the curl script now. Let it get to the point that it errors (the dependencies will have been installed) and then run the Python command again

adamsfour, 2023-02-12:
will do stand by

adamsfour, 2023-02-12:
How long should it run. Been running now for about 10 min. no errors

adamsfour, 2023-02-13:
I afraid still no luck.  Kivy did finally install. 
Steps I followed
Clean install of OS on SD card
Ran Curl command first
Ran Kivy install 
[ScreendumpKivyInstall.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10717681/ScreendumpKivyInstall.txt)

Ran Curl command again
[ScreenDumpcurl.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10717679/ScreenDumpcurl.txt)

adamsfour, 2023-02-13:
I was looking at closed cases to see if this had ever happen before. If you look at closed case #61. He had a similar issue. somehow he was able to get it to install. I know ultimately he had other issues but it does appear he was able to get it to work. It appears he using a PI 3 b+ vs me using a PI 3 b. thanks

peted-davis, 2023-02-13:
Sorry, I made a silly mistake in the Python command I shared. It installed v2.1.0 of Kivy, while the `curl` script tries to install v2.0.0. The `curl` script will try and overwrite the existing version, hence why it failed again. Run this command (it will take a long time to compile again)

`python3 -m pip install https://kivy.org/downloads/2.0.0/Kivy-2.0.0.tar.gz`

and then run the `curl` script for the last time. Hopefully this time it'll work

adamsfour, 2023-02-13:
no problem. I will give it a shot. I always go back and re-image the SD card first so I know I am starting with a clean build. Will report back once completed

peted-davis, 2023-02-13:
Ok, make sure you run the `curl` script first before the `python` command on the clean build

adamsfour, 2023-02-13:
yep

peted-davis, 2023-02-13:
This is annoying! It looks like I need to bump the version number as something has gone awry with v2.0.0. Can you trying running this command (no need for a clean install):

`python3 -m pip install https://github.com/kivy/kivy/archive/2.1.0.zip`

If that installs we have a solution. Don't try and run the `curl` script afterwards. I will need to bump the version number in the Github repo

adamsfour, 2023-02-13:
I kick off the script before I had to run out for a few hours. Yesterday I actually had to restart the install as it seem to stall after about two hours

adamsfour, 2023-02-13:
Also second time I ran it I increased swap space

adamsfour, 2023-02-13:
Sorry for the delayed responds. Let me know when you have the new script ready and I will give it a shot

 The install worked find for the new version Kivy. Screen dump attached 
[ScreenDumpKivyinstall.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10723559/ScreenDumpKivyinstall.txt)

peted-davis, 2023-02-13:
OK, I've bumped the version number in the `curl` script. Can you run it and let me know whether everything completes and you're able to start the console

adamsfour, 2023-02-13:
Will do. Just started script

adamsfour, 2023-02-13:
Bingo, Up and running perfectly.  Thanks for the help

closed 2023-02-13 by peted-davis