peted-davis/WeatherFlow_PiConsole #87: WFPIConsole buttons will not work when touching screen. Must use mouse to click button for it to activate

creas002, 2023-02-13:
**Issue:** 
WFPIConsole buttons will not work when touching the touch screen. It worked ~6 months ago but broke after a WFPIConsole update which required significant updates to buster OS . I must use mouse to click one of the WFPIConsole buttons.

**To Reproduce**

Tap button with finger on touch screen. For example, with the webbroswer launched I can touch screen to select a link or shortcut and it works correctly. When WFPIConsole is running, touching the screen will move the mouse pointer around but will not actually perform a click event for selecting that button. Buttons being Sager, Forecast, Lightning, Solar, Moon, Wind Speed, Menu, 

**Hardware/OS/Version**

 - OS: Raspbian GNU/Linux 10 (buster)
 - Hardware: Rpi 4b 4GB
 - Console Version: v22.12.3

[wfpiconsole_update.log](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10724917/wfpiconsole_update.log)
[wfpiconsole_20230213.log](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10724951/wfpiconsole_20230213.log)

peted-davis, 2023-02-13:
When did this stop working? There was an update to the code a while back that changed this behaviour based on the setting of `Cursor` in the `[Display]` section of the `wfpiconsole.ini` file. In order to fix a bug I have to disable touch input if the cursor is enabled. Try changing the value of `Cursor` to 0 and see if that fixes the issue

creas002, 2023-02-13:
Perfect, That was the issue.  It's working correctly now.  Thank you

[Display]
Cursor = 0