peted-davis/WeatherFlow_PiConsole #88: Oberservation from Bug report #86

adamsfour, 2023-02-13:
Overall it seems to be working but I am seeing some unexpected features/stability difference from my original install
First some background. Original install is running on a PI 4 and using old Kivy library. But up and running for well over a month with zero issues. I have it running on a always on display and it work perfectly 

New setup is running on a PI 3 and new Kivy library 

These are observation for the new install not seen on old install 

1) I can't access the setup menu.   I hit menu and then setup I get a blank screen no option nothing shows up 
2) I turn on autostart and if I choose to exit the program if would re-start on its own (is that expected). 
3) every so often the screen just starts to flash for a minute and then re-stabilize.  (see screen shot)
4) This is a strange one. If open a virtual desktop session (VNC) from my desktop I don't see the display I see the Linux main screen. even when the software is up and running.  I do have VNC enabled. If I VNC to my original setup I do get the weather display screen. . Is this some sort of setting I am missing in the OS?

I have attached two screen shot. one show the Virtual connection on the left and the direct HTML connection to the PI. This is totally different than the original where I get the same display on virtual and direct. 

The second picture show the stability issue. Notice the missing graphic on the wind speed. If comes and goes.

I also attached Attached is the runtime message I am see on the screen. I see some warnings
[runtimescreen.txt](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10725484/runtimescreen.txt)

![IMG_1867](https://
[image]
user-images.githubusercontent.com/122233494/218553532-888088c2-9a22-4526-b1dd-1d95ed2a515d.jpg)

Thanks for all the help

adamsfour, 2023-02-13:
[image]

peted-davis, 2023-02-13:
- I can't access the setup menu. I hit menu and then setup I get a blank screen no option nothing shows up
- every so often the screen just starts to flash for a minute and then re-stabilize. (see screen shot)

These two are related and sporadically occur on some Pi3 setups due to the amount of memory available to the GPU. They can be fixed by opening a terminal and running `sudo raspi-config` -> `Performance Options` -> `GPU Memory` and increasing the value to 256. Reboot once you have made this change.

- I turn on autostart and if I choose to exit the program if would re-start on its own (is that expected).

This is a 'feature'. It is designed such that users running without mouse or keyboard can recover from a major console crash more easily. You can permanently disable auto start or just use `wfpiconsole stop` to stop the console for the current session

- This is a strange one. If open a virtual desktop session (VNC) from my desktop I don't see the display I see the Linux main screen. even when the software is up and running. I do have VNC enabled. If I VNC to my original setup I do get the weather display screen. . Is this some sort of setting I am missing in the OS?

This is a known limitation of the Pi3. It has much weaker graphics hardware than a Pi4, so the console is rendered entirely differently. The result is that it is not visible when connecting via VNC.

adamsfour, 2023-02-13:
Outstanding thanks

closed 2023-02-14 by peted-davis