peted-davis/WeatherFlow_PiConsole #89: Won't start after kivx update.

pjmpessers, 2023-02-13:
**Describe the bug**
A clear and concise description of what the bug is.
 After update the screen ent black and then comes back with the terminal screen. Removed wfpiconsole with "rm -rf ~/wfpiconsole/" and installed with "curl -sSL https://peted-davis.github.io/wfpiconsole | bash".
Can i scroll back to the old kivx?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.
pi@raspberrypiweerstation:~ $ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_23-02-13_18.txt
[INFO   ] [Kivy        ] v2.1.0
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.9/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[INFO   ] [Config      ] Verifying station details
[INFO   ] [Factory     ] 189 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <sdl2>
[INFO   ] [GL          ] OpenGL version <b'2.1 Mesa 20.3.5'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'V3D 4.2'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 1
[INFO   ] [GL          ] Shading version <b'1.20'>
[INFO   ] [GL          ] Texture max size <4096>
[INFO   ] [GL          ] Texture max units <16>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [Websocket   ] 2023-02-13 23:41:42 - Opening connection
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event1
[INFO   ] [HIDInput    ] Read event from </dev/input/event1>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [HIDMotionEvent] using <vc4-hdmi-1>
[INFO   ] [HIDMotionEvent] using <vc4-hdmi-0>
[INFO   ] [Websocket   ] 2023-02-13 23:41:43 - Connection open

**Hardware/OS/Version (please complete the following information):**
 - OS: [e.g. Raspberry Pi OS Bullseye] Raspberry PI Bulseye debian
 - Hardware: [e.g. Pi3] Ri Model 4b
 - Console Version: [e.g. 22.3.1] 
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

**Additional context**
Add any other context about the problem here.

peted-davis, 2023-02-14:
OK, this is odd. Nothing in the log you shared suggests that Kivy is generating an error.

Can you try downgrading using this command and let me know how you get on:

`python3 -m pip install kivy[base]==2.0.0`

peted-davis, 2023-02-14:
Are you running the 32 bit or 64 bit version of Raspberry Pi OS?

pjmpessers, 2023-02-14:
That will be the 64 bits version.

peted-davis, 2023-02-14:
OK, that's the issue. The 64 bit version is not compatible at the moment. I suggest reverting to the 32 bit version of Bullseye and I will update the README to make it clear that the 64 bit version is currently not compatible

pjmpessers, 2023-02-14:
Aaaaaaah. That could be the case. Glad to know what the issue was. Thanks again.

closed 2023-02-21 by peted-davis