peted-davis/WeatherFlow_PiConsole #90: Will not load Python package

john0500, 2023-02-18:
I could not preform upgrade due to error. You suggested starting fresh install. I did the steps you outlined and it starts installs and fails at installing cryptographic.

I tried updating pip and came to already meets requirements.

peted-davis, 2023-02-21:
I need to know more information to help fix this. Can you let me know exactly what hardware you are using, and operating system you are running (e.g. Raspberry Pi OS Buster or Bullseye, 32 bit or 64 bit, or something else entirely)

john0500, 2023-02-21:
Hello, hardware is a Pi 4 Model B, The OS Raspberry, version not known , sorry windows guy. If you can tell me where to look I will. it is 32 Bit

peted-davis, 2023-02-21:
Ok, thanks for the info. Assuming nothing else is running on that Pi, can you try again with a fresh install of Raspberry Pi OS 32 bit Bullseye (rather than Buster). I think it is Buster that is causing the compatibility issue

john0500, 2023-02-21:
what's the best way to make sure I get the Bullseye install

peted-davis, 2023-02-21:
You can either select it when flashing the SD card using the Raspberry Pi Imager, or you can manually download a copy of the OS here: https://www.raspberrypi.com/software/operating-systems/ (it's the first option) and flash it to the SD card manually using the Raspberry Pi Imager. There's some more documentation here: https://www.raspberrypi.com/documentation/computers/getting-started.html#installing-the-operating-system

john0500, 2023-02-22:
hello, that addressed the issue . I have one small issue. For some reason it did now like my indoor temperature and it come up with a blank reading on indoor temp. Is that easy to fix. Thanks for the help.

john0500, 2023-02-22:
All back to original and updated. Edited the INI file and good to go.

Thanks again.

closed 2023-03-03 by peted-davis