peted-davis/WeatherFlow_PiConsole #91: KeyError: 'station_units'

letizido, 2023-03-02:
I have been using wfpiconsole for several years, and just discovered 64-bit Raspbian is no longer supported. So I re-imaged my Raspberry Pi 4B and tried to reinstall wfpiconsole according to instructions. I am encountering the following error during the configuration phase, just after entering 'n' on the "do you have an air" question. Here is the traceback error. I renewed the access token and verified the CheckWX API key:

  Observation units
  ---------------------------------
 Traceback (most recent call last):
   File "/home/dominic/wfpiconsole/main.py", line 33, in <module>
     configFile.create()
   File "/home/dominic/wfpiconsole/lib/config.py", line 97, in create
     writeConfigKey(Config, Section, Key, default[Section][Key])
   File "/home/dominic/wfpiconsole/lib/config.py", line 501, in writeConfigKey
     Value = OBSERVATION['station_units']['units_' + Key.lower()]
 KeyError: 'station_units'

letizido, 2023-03-02:
Raspbian bullseye 32-bit

pjmpessers, 2023-03-02:
Can be a server problem from weatherflow server of some kind.
I saw no update on my wfpiconsole. So i did a restart. After that i get a message:"Station id did not respond" or something like that. I'm not home, so i don't know what the message was exactly.

letizido, 2023-03-02:
My wfpiconsole start command is failing to write a wfpiconsole.ini file, from what it looks like. Just need to find out why.

rleeden, 2023-03-03:
It seems that something has been updated with the tempest API, and it is no longer accepting the legacy `api_key` as a URL parameter when being called. I discovered this when my wfpiconsole was failing to restart. See my forum post [here](https://community.weatherflow.com/t/error-starting-wfpiconsole-unable-to-fetch-station-metadata/20254?u=rleeden)

Essentially in file ~`pi/wfpiconsole/lib/config.py`, replace `api_key` with `token` on lines 188, 401 & 557 and hopefully that will fix things for you.

pjmpessers, 2023-03-03:
Aaaaah. Problem solved. Thank you, Richard.

bgiuriceo, 2023-03-03:
Thanks Richard...it's working again ...thanks for the rapid repair!!

peted-davis, 2023-03-03:
Sorry for this issue. I have just released a new version (23.3.1) that should fix the problem

wlwakefield, 2023-03-03:
> Sorry for this issue. I have just released a new version (23.3.1) that should fix the problem

Appreciate the fast fix, as always, however, when I run wfpiconsole update I get the following error this morning:

Error: Unable to update Python module: cryptography

 Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cryptography==38.0.1
  Using cached cryptography-38.0.1.tar.gz (599 kB)

Running the suggested actions further down the error don't result in success, unfortunately.

peted-davis, 2023-03-03:
My guess is that you are trying to run the console on a Pi4, but are using the older Buster version of Raspberry Pi OS. Unfortunately you are going to have to update to the latest Bullseye version to solve these compatibility issues

closed 2023-03-05 by peted-davis