peted-davis/WeatherFlow_PiConsole #92: Right Border Cut Off

letizido, 2023-03-03:
The right side border is missing when I run wfpiconsole in full-screen on my OEM 7" Pi Display.

When I start wfpiconsole in full screen, the right side border is missing from the image on my Raspberry Pi display.

[image]

**Hardware/OS/Version (please complete the following information):**
 - OS: Raspberry Pi OS Bullseye]
 - Hardware: Pi 4B
 - Console Version: 23.3.1

peted-davis, 2023-03-04:
Is this a brand new install or has it happened after an update? Do you see the same issue when running other programmes on the Pi

ctgreybeard, 2023-03-04:
I saw the same thing on my new Bullseye install but it seems to have corrected itself. I think it showed when I started the console from the commandline but is OK when auto started? I’m not sure …

peted-davis, 2023-03-04:
Ok, I will try and have a look into this.

letizido, 2023-03-04:
Thanks for the tips; I uninstalled and reinstalled wfpiconsole, and the first time it started without autostart-enable, the border was missing. I set autostart-enable and rebooted, and the border is displaying again. Must have something to do with initial startup.

peted-davis, 2023-03-05:
OK, glad you're up and running. I'll close this for the time being, but let me know if the error comes back

closed 2023-03-05 by peted-davis