peted-davis/WeatherFlow_PiConsole #93: Issues installing the update from yesterday 3/3/23

donaldnaegele, 2023-03-04:
I'm kinda new to the Pi. I only use it for the weather console. It's been working for a year or so with no issues.

I updated the Pi software today and then tried to update WeatherFlow and I'm getting 2 errors. I'm SO lost.

Suggestions?

[image]
[image]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: [e.g. Raspberry Pi OS Bullseye]
 - Hardware: [e.g. Pi3]
 - Console Version: [e.g. 22.3.1]

**Additional context**
Add any other context about the problem here.

peted-davis, 2023-03-04:
I need to know what version of the Raspberry Pi you are using (3 or 4) and what version of Raspberry Pi OS you are using (Buster or Bullseye). You can find that out by opening a terminal and running `cat /etc/os-release`

donaldnaegele, 2023-03-04:
Is this what you need? I don't know if I have a 3 or 4. How can I tell? It's in this big case where I can't see any text on it.

Don
[image]

peted-davis, 2023-03-04:
Can you show me the output of `cat /proc/device-tree/model`

donaldnaegele, 2023-03-04:
I think this is the version.

Don
[image]

peted-davis, 2023-03-04:
Yep, got it. 

The problem you're having is that you have a Pi4 but you are running the legacy (Buster) version of Raspberry Pi OS. This use to be a compatible combination, but this update now requires the most recent version of Raspberry Pi OS (Bullseye). I suggest you start with a fresh install of the OS on your SD card and install the console from scratch. This will fix the issues you are seeing

donaldnaegele, 2023-03-04:
Lots of work. But it’s up and running now. Thank so much for your help. Case closed.

closed 2023-03-05 by peted-davis