peted-davis/WeatherFlow_PiConsole #94: update install on wfpiconsole

rwb207, 2023-03-05:
pi@raspberrypi:~ $ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_23-03-05_24.txt
[INFO   ] [Kivy        ] v2.1.0
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.7/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Oct 31 2022, 14:04:00) 
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!

  ===================================================
  New version detected                               
  Starting wfpiconsole configuration wizard          
  ===================================================

  Required fields are marked with an asterix (*)     

  API keys
  ---------------------------------
 Traceback (most recent call last):
   File "main.py", line 35, in <module>
     configFile.update()
   File "/home/pi/wfpiconsole/lib/config.py", line 155, in update
     copyConfigKey(newConfig, currentConfig, Section, Key, defaultConfig()[Section][Key])
   File "/home/pi/wfpiconsole/lib/config.py", line 270, in copyConfigKey
     validateAPIKeys(newConfig)
   File "/home/pi/wfpiconsole/lib/config.py", line 531, in validateAPIKeys
     CHECKWX = requests.get(URL, headers=header).json()
   File "/usr/lib/python3/dist-packages/requests/api.py", line 75, in get
     return request('get', url, params=params, **kwargs)
   File "/usr/lib/python3/dist-packages/requests/api.py", line 60, in request
     return session.request(method=method, url=url, **kwargs)
   File "/usr/lib/python3/dist-packages/requests/sessions.py", line 533, in request
     resp = self.send(prep, **send_kwargs)
   File "/usr/lib/python3/dist-packages/requests/sessions.py", line 646, in send
     r = adapter.send(request, **kwargs)
   File "/usr/lib/python3/dist-packages/requests/adapters.py", line 449, in send
     timeout=timeout
   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
     chunked=chunked)
   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 343, in _make_request
     self._validate_conn(conn)
   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 841, in _validate_conn
     conn.connect()
   File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 332, in connect
     cert_reqs=resolve_cert_reqs(self.cert_reqs),
   File "/usr/lib/python3/dist-packages/urllib3/util/ssl_.py", line 264, in create_urllib3_context
     context = SSLContext(ssl_version or ssl.PROTOCOL_SSLv23)
   File "/usr/lib/python3/dist-packages/urllib3/contrib/pyopenssl.py", line 391, in __init__
     self._ctx = OpenSSL.SSL.Context(self.protocol)
   File "/home/pi/.local/lib/python3.7/site-packages/OpenSSL/SSL.py", line 674, in __init__
     res = _lib.SSL_CTX_set_ecdh_auto(context, 1)
 AttributeError: module 'lib' has no attribute 'SSL_CTX_set_ecdh_auto'

4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: [e.g. Raspberry Pi OS Bullseye]
 - Hardware: [e.g. Pi3]
 - Console Version: [e.g. 22.3.1]

**Additional context**
Add any other context about the problem here.

This is what I get after running update and trying to restart

peted-davis, 2023-03-05:
Looks like you are running the legacy Buster version of Raspberry Pi OS on a Pi 4. You need to upgrade to the latest 32 bit Bullseye version of Raspberry Pi OS

rwb207, 2023-03-05:
Ok thank you

Ray Barton
908.309.2222

closed 2023-03-08 by peted-davis