peted-davis/WeatherFlow_PiConsole #95: wfpiconsule update fails [✗] Hardware check failed (aarch64)

w9dls, 2023-03-05:
Tried updating wfpiconsole from v22.12.3 to v23.3.1. It fails on both of my Raspberry Pi 4s.

cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

[image]

peted-davis, 2023-03-05:
The installer cannot be used on the 64 bit version of Raspberry Pi OS. I suggest installing the 32 bit version and then you will be able to install the latest version

disruptivepatternmaterial, 2023-03-12:
@peted-davis can you elaborate? Would be great to fix this...happy to take a look. Unless it is an issue in a dependancy.

w9dls, 2023-03-12:
Pete, I was able to install the latest version with no apparent issues by adding aarch* as an acceptable type of CPU on the hardware check. 
I may have done this before as I had the earlier versions loaded on my Pi 4s with the 64 but RPi OS. 

Don Stedman.

closed 2023-04-29 by peted-davis