peted-davis/WeatherFlow_PiConsole #96: Screen Blanking Not Working in Bullseye

letizido, 2023-03-07:
Using 32-bit Bullseye on a Pi4B. wfpiconsole installs and runs, but there appears to be a bug in Bullseye that breaks screen blanking on the Pi touchscreen. I installed 32-bit Buster, and screen blanking works properly in the OS. Then, when I tried installing wfpiconsole, I encountered several install failures around the cryptography module. Installer is complaining about not being able to build the wheel for cryptography and needing Rust, and will not complete. So at this point it looks like a choice between wfpiconsole with no screen blanking or screen blanking without wfpiconsole. I also tried xscreensaver but it will only blank the data on the screen. It doesn't turn off the backlight.

ctgreybeard, 2023-03-08:
I actually turned off screen blanking using raspi-config when I upgraded to bullseye. But I'm not sure if it turned off the backlight. I also seem to recall that there is a boot config option to enable it. I do have that set on my homebridge server and can dig it out if you like. It blanks the text console though, maybe not the graphical one.

letizido, 2023-03-09:
I resolved this issue by downgrading the xserver-xorg-core package with the below command:

apt install xserver-xorg-core=2:1.20.11-1+deb11u5

There is apparently a bug in the newxserver-xorg-core that breaks screen blanking in Bullseye.

closed 2024-10-24 by peted-davis