peted-davis/WeatherFlow_PiConsole #97: Any Interest in a Module to Support AHT20 for Inside Temperature?

letizido, 2023-03-10:
Firstly, my apologies for opening an issue for what is clearly a feature request.

I have been happily using wfpiconsole for several years. I got to wondering how to use the "Inside Temperature" feature, and understand that it was written for the Weatherflow Air device, which is no longer being sold since Tempest was released. Weatherflow also seems not to be actively developing the "Breathe" product either.

Which leads to my request. I was able to spend around $10 to acquire an Adafruit AHT20 along with a QWIIC shim and cable, and with a simple python script I can now read temperature and humidity.

Is there any interest in developing a module to get temperature (and potentially humidity) and insert them into the Temperature panel on wfpiconsole? My python skills are not sufficient to do this myself, but I would be happy to test.

It seems like a really nice and inexpensive way to add Indoor temperature to wfpiconsole in lieu of the old WF Air module not being available any more.

ctgreybeard, 2023-03-10:
I actually did it once and still have the AHT connected but it went away during an upgrade and I didn’t resurrect it. What would be nice would be to have a plug-in interface of some sort.

letizido, 2023-03-10:
Would you be willing to share your code? If I can find someone to get it working again I will contribute it back, hopefully as a plug-in.

ctgreybeard, 2023-03-10:
Willing, sure. Find it? Hmmmm … I’ll see if I can dig it up.

disruptivepatternmaterial, 2023-03-12:
Yeah I would love to have something like this. I have AQI, VOC, CO2, CO, etc... sensors in my home based on ESPHome and home-assistant. Ideally, a panel that can reference something like a list of home-assistant entities would be rad, but the framework used on this project is very unfamiliar to me, so I never got to it....

letizido, 2023-03-24:
Hello ctgreybeard,

Were you able to locate the AHT code you wrote to integrate indoor temp into wfpiconsole?

ctgreybeard, 2023-03-24:
Sorry, no. I have not come across it. And my project list is pretty full but I don’t recall it was that difficult. I think I first followed the example on Adafruit to get the readings then reverse-engineered it into the console.

—
Bill Waggoner
Sent with Spark

letizido, 2023-04-19:
Peter, if it helps, I would send you an AHT20 and QWIIC socket in the hope that you could develop an integration for the AHT20 for indoor temperature/humidity. I have reviewed the Weatherflow forum, and it seems that the indoor temperature/humidity topic has come up several times over the years.

ctgreybeard, 2023-05-06:
I have the AHT20 basic code running but not yet in wfpiconsole, just the example code: [Chouffy / python_sensor_aht20](https://github.com/Chouffy/python_sensor_aht20) and [adafruit / Adafruit_CircuitPython_AHTx0](https://github.com/adafruit/Adafruit_CircuitPython_AHTx0) both work fine. One thing I had to do was run `raspi-config` and **enable i2c**.

I am wondering what the best approach to integrating it into the wfpiconsole would be. Any thoughts?

Here's how I have it mounted. I printed the bracket that attaches via the holes in the cover.

[image]
[image]

peted-davis, 2023-05-07:
> I am wondering what the best approach to integrating it into the wfpiconsole would be. Any thoughts?

Best to incorporate it through a custom panel. That way it won't be overwritten whenever the console updates. 

You'll want to turn the code you have to read the sensor into a function, and then you can call that function from the console using `Clock.schedule_interval(...)‘. Have a look in `main.py` to see how I use the clock function to schedule updates to the forecast, sunrise/sunset and moon phase. You'll then just need to have your function update the variables that you use to display the temperature/humidity. These variables should be defined in your custom panel.

ctgreybeard, 2023-05-07:
I looked at the current temperature panel and there is a lot of logic around IndoorTemp that would need to be duplicated in a new panel. 

My thought was to inject IndoorTemp into the data flow as if it had come in via the sensors. That way the current panel could display it as is.

peted-davis, 2023-05-09:
Yes, this is perfectly possible - you'll just need to be very careful when downloading a new update as by default `wfpiconsole update` will overwrite any changes you have made that are not in the `user/` folder.

If you want to inject the values into the current temperature panel, then you just need to set the value of `app.CurrentConditions.Obs[inTemp]`, `app.CurrentConditions.Obs[inTempMax]`, and `app.CurrentConditions.Obs[inTempMin]`. You can follow the logic in `lib/observationParser.py`

DMEbner, 2023-11-09:
Hello! Do you have a set of instructions on how to import different indoor weather sensors from several different companies?

overlind, 2024-04-21:
Initially, I was going to try to integrate an AHT20 to report indoor temperature and humidity into the existing temperature panel by forking the project to avoid accidental upgrades.  However, after reading the recommendation by @peted-davis to create a custom panel, I found this was much simpler than trying to inject the data into an existing panel.  I duplicated the original temperature panel as a custom one, replacing the "Feels Like" section with the indoor readings.  I also brightened the current outdoor/indoor temperatures to help call attention to them.

As a prerequisite, I had to run the following two commands from the commandline on my RaspberryPi 4 (64-bit) after installing wfpiconsole based on the normal installation instructions.  Some of those packages may have already been installed.

```
source ~/wfpiconsole/venv/bin/activate
pip install board adafruit-circuitpython-dht RPi.GPIO
```

The attached [customPanels.zip](https://github.com/peted-davis/WeatherFlow_PiConsole/files/15050455/customPanels.zip) contents should go in the   `~/wfpiconsole/user/`  folder.  The self.sensor_pin in the .py file needs to be updated to use the correct data pin for your AHT20 if using something other than pin 4.  The custom panel should show up as "TemperatureRP" in the menu of WeatherFlow_PiConsole.

I am brand new to working with a Raspberry Pi, so I add the GPIO.cleanup() because the sensor sometimes became unresponsive in my testing.  I think it could reset additional components also connected to the GPIO pins as a side effect.  I had a difficult time using local class variables from the Kivy panel, so I ended up extending the Obs dictionary and adding error checking to avoid undefined keys.

[image]