peted-davis/WeatherFlow_PiConsole #99: Update fails with: Unrecognised Raspberry Pi (Raspberry Pi 2 Model B Rev 1.1)

PideaLtd, 2023-03-15:
When updating to the latest version I get the error:

"Unrecognised Raspberry Pi (Raspberry Pi 2 Model B Rev 1.1)"

Here's the device info:

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

 cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                                                                                                              vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                                                                                                              vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                                                                                                              vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                                                                                                              vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a01041
Serial          : 000000008dae4aa1
Model           : Raspberry Pi 2 Model B Rev 1.1

peted-davis, 2023-03-17:
Yes, this is an unsupported Raspberry Pi model. A Raspberry Pi 3 or greater is required to use the installer

peted-davis, 2023-03-17:
> Oh, what is the reason behind dropping support for older versions of the Raspberry Pi that have previously been supported ?

It is important to note that these older systems have never 'officially' been supported. However, I did allow the installer to run on these systems with the understanding that support would be provided on a 'best efforts' basis. Despite this, I have had numerous cases of people expecting full support despite running unsupported hardware. 

I don't have access to unsupported hardware and therefore given the time I was spending having to solve issues generated by this 'ad hoc' support, I took the decision to stop the installer running on these systems. I know it is frustrating, but I support this project entirely in my spare time for no monetary gain, so I have to make these decisions.

> Is there a way to suppress the update available notification as I'm not going to update the hardware ?

Not at the moment, but I will be bringing this functionality soon

PideaLtd, 2023-03-17:
Thanks for the reply.  Would it be possible to add the RPi 2 as supported hardware ?  It's been running perfectly fine without any issues for the last couple of years on mine.  If it would help I've got a spare RPi 2 here that I could donate to you for testing if you wish.

peted-davis, 2023-03-17:
OK, I have made some tweaks to the installer that should allow it to run on a Raspberry Pi 2. At the same time, I have added a clear warning that this is an unsupported systems, so errors might occur. As you have been running for a couple of years, this seems unlikely at the moment at least. Can you try running `wfpiconsole update` again?

PideaLtd, 2023-03-17:
That's awesome ... much appreciated.

TL;DR - it works perfectly !

As requested I ran wfpiconsole update and got the following:

$ wfpiconsole update

  [✓] Root user check passed
  [✓] Architecture check passed (armv7l)
  [✓] OS check passed (Raspbian GNU/Linux 10 (buster))
  [!] Raspberry Pi check warning (Raspberry Pi 2 Model B Rev 1.1)

  [!] WARNING: unsupported architecture or Raspberry Pi detected
      No support is available for errors encountered while running
      the PiConsole

  ==============================
  Updating WeatherFlow PiConsole
  ==============================

  [✓] Checking for updated packages
  [i] 209 updated packages available. Use 'sudo apt upgrade' to install

  [i] WeatherFlow PiConsole dependency checks...
  [✓] Checking for git
  [✓] Checking for curl
  [✓] Checking for rng-tools
  [✓] Checking for build-essential
  [✓] Checking for python3-dev
  [✓] Checking for python3-pip
  [✓] Checking for python3-setuptools
  [✓] Checking for libssl-dev
  [✓] Checking for libffi-dev
  [✓] Checking for libatlas-base-dev
  [✓] Checking for jq

  [i] Updating WeatherFlow PiConsole Python modules...
  [✓] Updating Python package manager

  [i] Kivy Python library dependency checks...
  [✓] Checking for pkg-config
  [✓] Checking for libgl1-mesa-dev
  [✓] Checking for libgles2-mesa-dev
  [✓] Checking for libgstreamer1.0-dev
  [✓] Checking for gstreamer1.0-plugins-bad
  [✓] Checking for gstreamer1.0-plugins-base
  [✓] Checking for gstreamer1.0-plugins-good
  [✓] Checking for gstreamer1.0-plugins-ugly
  [✓] Checking for gstreamer1.0-omx
  [✓] Checking for gstreamer1.0-alsa
  [✓] Checking for libmtdev-dev
  [✓] Checking for xclip
  [✓] Checking for xsel
  [✓] Checking for libjpeg-dev
  [✓] Checking for libsdl2-dev
  [✓] Checking for libsdl2-image-dev
  [✓] Checking for libsdl2-mixer-dev
  [✓] Checking for libsdl2-ttf-dev

  [i] Kivy Python library installation check (will be updated)
  [✓] Updating Kivy Python library
  [✓] Updating Kivy configuration for touch screen

  [i] Latest version of WeatherFlow PiConsole: v23.3.1
  [i] Installed version of WeatherFlow PiConsole: v22.12.3

  [✓] Updating WeatherFlow PiConsole to v23.3.1

  =============================================
  WeatherFlow PiConsole update complete!
  Restart the console with: 'wfpiconsole start'
  or 'wfpiconsole autostart-enable'
  =============================================

I then started the console with 'wfpiconsole start':

$ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_23-03-17_5.txt
[INFO   ] [Kivy        ] v2.1.0
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.7/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[INFO   ] [Config      ] Verifying station details
[INFO   ] [Factory     ] 189 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: egl_rpi
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'OpenGL ES 2.0'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'VideoCore IV HW'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 0
[INFO   ] [GL          ] Shading version <b'OpenGL ES GLSL ES 1.00'>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <8>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [Websocket   ] 2023-03-17 17:43:47 - Opening connection
Unable to connect to X server
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [MTD         ] Read event from </dev/input/event0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [MTD         ] </dev/input/event0> range position X is 0 - 799
[INFO   ] [MTD         ] </dev/input/event0> range position Y is 0 - 479
[INFO   ] [MTD         ] </dev/input/event0> range touch major is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range touch minor is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range pressure is 0 - 255
[INFO   ] [MTD         ] </dev/input/event0> axes invertion: X is 0, Y is 0
[INFO   ] [HIDMotionEvent] using <raspberrypi-ts>
[INFO   ] [MTD         ] </dev/input/event0> rotation set to 0
[INFO   ] [HIDMotionEvent] <raspberrypi-ts> range ABS X position is 0 - 799
[INFO   ] [HIDMotionEvent] <raspberrypi-ts> range ABS Y position is 0 - 479
[INFO   ] [HIDMotionEvent] <raspberrypi-ts> range position X is 0 - 799
[INFO   ] [HIDMotionEvent] <raspberrypi-ts> range position Y is 0 - 479
[INFO   ] [Websocket   ] 2023-03-17 17:43:48 - Connection open

So went ahead and did Ctrl C to escape and did:

$ wfpiconsole autostart-enable

  [✓] Root user check passed

  ======================================
  Enabling console autostart during boot
  ======================================

  [✓] Copying service file to autostart directory
  [✓] Enabling WeatherFlow PiConsole service file

  ====================================================
  WeatherFlow PiConsole autostart sucesfully enabled
  Console will now start automatically at boot up
  Starting console for current session. Please wait...

Everything looks fine so I went ahead and did sudo apt-get update && sudo apt-get upgrade -y as I've got a lot of old packages and then rebooted the RPi and it's all still working.

Thanks for making those changes.

closed 2023-03-18 by peted-davis