peted-davis/WeatherFlow_PiConsole #100: Update fails based on "Error: Failed building wheel for cryptography"

NickT-in-SD, 2023-03-16:
I am currently on version 22.12.3 of the WeatherFlow Pi Console, and it is now constantly blanking the screen and asking me to update to 23.3.1. However, when I try that update, I get "Error: Failed building wheel for cryptography". I have updated my Pi OS and as far as I can tell, it passes all the tests when it first boots up. Not sure what to do.

peted-davis, 2023-03-17:
Are you running the legacy (Buster) version of Raspberry Pi OS on a Pi4? If so, you need to upgrade to the latest Bullseye version

NickT-in-SD, 2023-03-17:
Yes, I believe I am. I’ll have to research the upgrade steps for moving to Bullseye. Thank you.

peted-davis, 2023-03-17:
Best to start with a brand new image flashed to your SD cards. In-place upgrades cause a lot of problems

NickT-in-SD, 2023-03-18:
Loaded a new image and reinstalled the console. Working fine now, thanks!

Would love to get the Ecobee integration for my home.

closed 2023-03-18 by peted-davis