peted-davis/WeatherFlow_PiConsole #102: Data not displaying any more

sublimeshutter, 2023-04-13:
The 4 panes Temp, Wind Speed, Rainfall and Barometer are not displaying data from my tempest. I have tried updating the WeatherFlow software and the Pi software. Have restarted several times. checked the API as well.

Any suggestions on what to try next?

Cheers,

peted-davis, 2023-04-14:
What is your station ID?

sublimeshutter, 2023-04-14:
ST-00021197

sublimeshutter, 2023-04-14:
> That's your Tempest serial number. I need to know the station ID. It's the number that appears in the web URL to your station page. E.g: https://tempestwx.com/station/33581/

Apologies

https://tempestwx.com/station/35430/grid

sublimeshutter, 2023-04-14:
> Linux tempest 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l
>
> The programs included with the Debian GNU/Linux system are free software;
>
> the exact distribution terms for each program are described in the
>
> individual files in /usr/share/doc/*/copyright.
>
> Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
>
> permitted by applicable law.
>
> Last login: Wed Apr  5 16:55:29 2023
>
>
> Bookshelf  Documents  Music     Public     Videos
>
> Desktop    Downloads  Pictures  Templates  wfpiconsole
>
>
>
> atlas  kvlang   main.py    service          wfpiconsole.kv       wfpiconsole.sh
>
> fonts  lib      panels     user             wfpiconsole.log
>
> [... 136 lines cut ...]
>
> ^X Exit          ^R Read File     ^\ Replace       ^U Uncut Text    ^T To Spell      ^_ Go To Line    M-E Redo         M-6 Copy Text    ^Q Where Was     M-W Next         ^F Forward       ^▶ Next Word     ^E End           ^N Next Line     M-+ Scroll Down
>
>
> Thanks! I can't see any issues there (except your battery voltage is relatively low). Are any errors printed to the terminal when you run the console? Can you share the complete contents of your wfpiconsole.ini file? That way I can run your station and investigate further

Here you go. Thanks very much for investigating!

Last login: Wed Apr  5 08:26:02 on console

***@***.*** ~ % ssh ***@***.******@***.***>

peted-davis, 2023-04-14:
No problem! I have no issues running your station here - everything is displaying fine. Have you checked the network connection on your Pi? 

[image]

sublimeshutter, 2023-04-14:
I can ssh into it from my mac so I assume that means it’s on my network. And looking at my router page it shows the pi connected…

I just went into Raspi-Config and re-entered the wifi and password but still no joy.

Sorry to have bothered you. I am missing something here

[unnamed (1)]<https://user-images.githubusercontent.com/10545710/231979846-e67a559e-519d-4be0-b246-72752cc2b584.jpg>

peted-davis, 2023-04-14:
Can you ping external websites from your pi? Try running `ping google.co.uk`

closed 2023-07-20 by peted-davis