peted-davis/WeatherFlow_PiConsole #106: wfpiconsole update issue

klockk, 2023-05-14:
Just noticed this today, trying to check for updates to the wfpiconsole code. It seems to be running using the current code, so not sure what changed recently?

pi@ADA:~/EdgeBridge $ sudo wfpiconsole update

  [✓] Root user check passed
  [!] Architecture check warning (aarch64)

  [✗] OS check failed (Raspbian GNU/Linux 10 (buster))

pi@ADA:~/EdgeBridge $ uname -a
Linux ADA 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux
pi@ADA:~/EdgeBridge $

peted-davis, 2023-07-20:
The install script is now much stricter when checking for compatible OS versions and architectures. Because you are running on an unsupported 64 bit architecture, you cannot using the Buster version of Raspberry Pi OS. I would suggest using the fully supported 32 bit version of Raspberry Pi OS on a Pi 4

closed 2023-07-20 by peted-davis

klockk, 2023-07-20:
I upgraded to Bullseye and that resolved the issue as well.