peted-davis/WeatherFlow_PiConsole #109: Place wfpiconsole on secundaire monitor. (Request)

pjmpessers, 2023-07-02:
I have a request that i can't find the answer for. 
I have connected a secundaire monitor on my raspberry pi and made a extended desktop. This was to show my sharptool dashboard. The strange thing is that the secundaire monitor has a misalignment with using the touchsreen. So i want to place the wfpiconsole on the secundaire monitor. But it always starts on the primaire monitor. Can i change this in the wfpiconsole.ini?

peted-davis, 2023-07-20:
You will have to stop the console running in full screen mode, and then you can move the console manually to the second screen

closed 2023-07-20 by peted-davis

pjmpessers, 2023-07-20:
And then run it there in full-screen? And is this a action that i need to repeat every time i reboot?

peted-davis, 2023-07-20:
No, once full screen is turned off, it will always run in windowed mode. I have no experience running the PiConsole on two monitors. I imagine it'll only ever go full screen on the primary monitor (not the secondary), so to get in on the secondary monitor you'll have to drag the window and then resize. You'll have to do this every time

pjmpessers, 2023-07-20:
Thanks for your help. I placed wfpiconsole on the small monitor, but this is secundair monitor and sharptools on the bigger primary monitor. So i will seek for a solution to make the small one primary.

Thanks again. 🙏🏻

Verzonden vanaf mijn Galaxy