peted-davis/WeatherFlow_PiConsole #110: Barometric pressure trend shows positive value when pressure is slowly falling

karlkec, 2023-07-19:
**Describe the bug**
When the barometric pressure trend is slowly falling,  the pressure change shows as a positive number.

**To Reproduce**
Steps to reproduce the behavior:
1. Set pressure units to inHg
2. Wait for barometric pressure to be slowly falling
3. Manually record pressure readings over an hour or two to confirm slowly falling
4. Note that pressure trend shows as a positive number.

**Hardware/OS/Version (please complete the following information):**
 - OS: Raspberry Pi OS Buster 32-bit
 - Hardware: Pi 3 model B, with 
 - Console Version: 23.5.beta

In lib/observationFormat.py line 258, I believe second argument to the round() function should be 3, not 1; to match the number of digits shown on the display for inHg/hr.
Similarly, in line 263, the second argument to round() should be 2 for mmHg/hr

peted-davis, 2023-07-20:
Good spot - I think the rounding issue you identified is the culprit. I will get this fixed up in the next version

peted-davis, 2023-10-26:
This is now fixed in v22.10.1

closed 2023-10-26 by peted-davis