peted-davis/WeatherFlow_PiConsole #112: WFPIconsole does not install on debian bookworm

pjmpessers, 2023-08-10:
**Describe the bug**
Try to install wfpiconsole on debrian bookworm

**To Reproduce**
Steps to reproduce the behavior:
1. Install with command (curl -sSL https://peted-davis.github.io/wfpiconsole | bash)

Getting error: 
[i] Installing WeatherFlow PiConsole Python modules...
  [✗] Updating Python package manager
  Error: Unable to update Python package manager: pip

 error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - debian bookworm
 - PI4
 

**Additional context**
Add any other context about the problem here.

peted-davis, 2023-08-11:
Thanks for the bug report. This is a known issue related to a change in how Python package managers work: https://peps.python.org/pep-0668/. I will update the installer to become compatible with this change

pjmpessers, 2023-08-11:
Thank you Pete.
Great to hear. 🙏🏻👍🏻

closed 2023-08-11 by pjmpessers