peted-davis/WeatherFlow_PiConsole #113: WFPICONSOLE goes offline.

pjmpessers, 2023-08-11:
**Describe the bug**
I have 2 wfpiconsole stations. Both have a raspberry pi 4B. On both the stations i have attatched 2 touchscreens. On 1 station it is a DSP with a HDMI monitor. On the other it is 2 HDMI touchscreens. The WFPIconsole is on the primary screen. On the other i let chromium show sharptools in kiosk mode. With sharptools i get information from my smarthome and controle it.

After a while (Can be a minute, can be hours, can be days) it goes offline. I see that when wind-direction, etc isn't being updated.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'menu'
2. Station status : "Offline."

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: Bullseye
 - RPI 4B 8 gb
 - Console version v23.5.beta and seen in previous.

**Additional context**
Happens with websocket and UDP.

[image]

peted-davis, 2023-08-11:
Ok, this might be hard to diagnose. Are any errors or warnings printed to the terminal when the console goes offline? When the console is offline, can you still access the internet using a web browser on your Pi? The fact it happens with both the Websocket and UDP suggests it is a network issue on the Pi

pjmpessers, 2023-08-11:
Thanks for your quick response, Pete. I understand this is and will be a difficult one.

Yes, Wifi is still up and running. I know this, because sharptools (The second screen) still gets all his information and still can controle my smarthome and gets live stream from my webcams.

pjmpessers, 2023-08-11:
Is there a possibility to log it somewhere?

pjmpessers, 2023-08-11:
[wfpiconsole.log](https://github.com/peted-davis/WeatherFlow_PiConsole/files/12320935/wfpiconsole.log)

I see it is setting up the touchscreen. But i also included xinput remap in /etc/xdg/lxsession/LXDE-pi/autostart
And i disabled autostart from wfpiconsole, because i got troubles with the second touchscreen where the cursor was way off.

So i included 

@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
@xset s off
@xset -dpms
@xset s noblank
@chromium-browser --window-position=1920,0 --kiosk https://sharptools.io/dashboard/view/ZrN77XH41ICKTwObatha?kiosk=true
@wfpiconsole start
xinput map-to-output 6 HDMI-2  

in the autostart

pjmpessers, 2023-08-11:
I saw a logging pass by. WFPIconsole wind-direction and speed was still changing every 6 seconds, but radiation and temperature where frozen. After a few minutes is saw it was still online, but the logging was saying:

 Exception in thread Websocket:
 Traceback (most recent call last):
   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
     self.run()
   File "/usr/lib/python3.9/threading.py", line 892, in run
     self._target(*self._args, **self._kwargs)
   File "/usr/lib/python3.9/runpy.py", line 268, in run_path
     return _run_module_code(code, init_globals, run_name,
   File "/usr/lib/python3.9/runpy.py", line 97, in _run_module_code
     _run_code(code, mod_globals, init_globals,
   File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
     exec(code, run_globals)
   File "service/websocket.py", line 310, in <module>
     loop.run_until_complete(main())
   File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
     return future.result()
   File "service/websocket.py", line 294, in main
     await asyncio.gather(*list(websocket.task_list.values()))
   File "service/websocket.py", line 265, in __async__listen
     await self.__async__watchdog()
   File "service/websocket.py", line 196, in __async__watchdog
     if self.config['System']['SkipRestObservations']:
   File "/usr/lib/python3.9/configparser.py", line 1254, in __getitem__
     raise KeyError(key)
 KeyError: 'SkipRestObservations'

pjmpessers, 2023-08-11:
After a few minutes it also stops the winddirection, enz and it goes offline. No new logging seen.
This can also be a no problem for days. In the weatherflow app it is still receiving data and reboot it is up and running again.

pjmpessers, 2023-08-12:
WoW. You are quick. Thank you. Installed the update. I let you know if it stays stable.

By the way, If i'm not mistaken this also applies to UDP.
And thanks again. Love the WFPIconsole.

pjmpessers, 2023-08-12:
Can i give a small donation?

peted-davis, 2023-08-12:
You're quick at spotting my changes! All your information was really useful - thanks! There are two issues going on here. First you are suffering from a known feature/issue/bug with the Websocket: https://community.weatherflow.com/t/websocket-api-issue-with-obs-st-messages/21078. I have a watchdog that resets the Websocket connection when this happens, but that had a bug in it. I've fixed that bug, so hopefully the watchdog will work correctly.

Can you keep an eye on the log when you see the same with the UDP. The UDP connection shouldn't suffer from the same issue as the Websocket and doesn't have a watchdog task, so the fix I made won't fix the UDP connection. 

 If you want to leave a small donation you can do that here, but don't feel like you have too! https://www.buymeacoffee.com/peted.davis

pjmpessers, 2023-08-12:
Sended you a small donation from what i hope is received.
I will try to do the same thing and hope to catch the same error logging like before.

It was still working, but didn't receive all the data as seen in my photo. I did receive this.

Sorry for putting in another problem. Should i make a new item? This is when using UDP.

[WARNING] [request_api ] 2023-08-12 17:31:31 - last_24h call failed
[WARNING] [request_api ] 2023-08-12 17:31:32 - Today call failed
[WARNING] [request_api ] 2023-08-12 17:31:32 - Yesterday call failed
[WARNING] [request_api ] 2023-08-12 17:31:33 - Month call failed
[WARNING] [request_api ] 2023-08-12 17:31:33 - Year call failed
[WARNING] [request_api ] 2023-08-12 17:32:31 - last_24h call failed
[WARNING] [request_api ] 2023-08-12 17:32:32 - Today call failed
[WARNING] [request_api ] 2023-08-12 17:32:32 - Yesterday call failed
[WARNING] [request_api ] 2023-08-12 17:32:33 - Month call failed
[WARNING] [request_api ] 2023-08-12 17:32:33 - Year call failed
[WARNING] [request_api ] 2023-08-12 17:33:31 - last_24h call failed
[WARNING] [request_api ] 2023-08-12 17:33:31 - Today call failed
[WARNING] [request_api ] 2023-08-12 17:33:32 - Yesterday call failed
[WARNING] [request_api ] 2023-08-12 17:33:33 - Month call failed
[WARNING] [request_api ] 2023-08-12 17:33:33 - Year call failed
[WARNING] [request_api ] 2023-08-12 17:34:31 - last_24h call failed
[WARNING] [request_api ] 2023-08-12 17:34:32 - Today call failed
[WARNING] [request_api ] 2023-08-12 17:34:32 - Yesterday call failed
[WARNING] [request_api ] 2023-08-12 17:34:32 - Month call failed
[WARNING] [request_api ] 2023-08-12 17:34:33 - Year call failed
[WARNING] [request_api ] 2023-08-12 17:35:31 - last_24h call failed
[WARNING] [request_api ] 2023-08-12 17:35:31 - Today call failed
[WARNING] [request_api ] 2023-08-12 17:35:32 - Yesterday call failed
[WARNING] [request_api ] 2023-08-12 17:35:32 - Month call failed
[WARNING] [request_api ] 2023-08-12 17:35:33 - Year call failed
[WARNING] [request_api ] 2023-08-12 17:36:31 - last_24h call failed
[WARNING] [request_api ] 2023-08-12 17:36:32 - Today call failed
[WARNING] [request_api ] 2023-08-12 17:36:32 - Yesterday call failed
[WARNING] [request_api ] 2023-08-12 17:36:33 - Month call failed
[WARNING] [request_api ] 2023-08-12 17:36:33 - Year call failed

[image]

pjmpessers, 2023-08-12:
P.S. Your link explained a lott. And a great idea that you let watchdog reconnect to websocket. Thanks.

peted-davis, 2023-08-13:
Your donation is gratefully received! 

Those warnings you are seeing indicate that requests to the WF API are failing. What is your `rest_api` system setting in the config file set to? If it is set to `1`, then there might be a network error occurring. The console is attempting to fetch data from the API, but it is failing. If `rest_api` is set to `0`, then these warnings are a bug as the console shouldn't be calling the API at all

pjmpessers, 2023-08-13:
I think you where right. Bot WFPIConsole where failing. Both system had the REST_API set to one. I did set both to 0 and it now collects more data without a error in the log.

But it looks it resetted the historical data, because a few things is resetted in UDP.

But a strange thing occured. After the first attempt it looked promising. But the second time it started to get errors again and REST_API was back to 1. This was on both system.

But it is not a problem for me. I just use websocket. If it is fixable take your time if you want to. Websocket is fine for me.

Picture is when it collected more data.

[image]

swvalenti194, 2024-02-18:
@pjmpessers @peted-davis Anyone care to share their watchdog config in order to reset the websocket?  I keep getting the same and not sure how to implement this fix with watchdog.

peted-davis, 2024-02-21:
@swvalenti194, the watchdog is already baked into the code. If your connection keeps dropping I would suggest investigating whether you are having network issues