peted-davis/WeatherFlow_PiConsole #116: Installer check for sudo user fails if installing from inside "sudo su - " shell

TanerH, 2023-08-27:
I believe you might need some more logic around checking EUID==0 and blindly using $SUDO_USER as the $USER variable, in the installer :)

I normally invoke all my root shells via "sudo su -" which will not have $SUDO_USER set, and so this check in your install script will fail.  I think you could either state something like "please run the installer from a user login" or perhaps you can just error out during install if "$SUDO_USER" ends up being blank, explaining you need to run this from a user shell, or from a "sudo bash" shell perhaps.

The result of this situation is that the install fails after trying to run the "usermod" utility, since $USER is blank (it just errors out and quits).

Otherwise, the installed worked just fine on my Ubuntu 22.04 box!

peted-davis, 2023-08-30:
OK, the installer was not designed to be called by an elevated user. I will make it exit with an error in this case in the future

peted-davis, 2023-10-26:
This has been implemented in v23.10.1

closed 2023-10-26 by peted-davis