peted-davis/WeatherFlow_PiConsole #117: Piconsole not starting

TimoHav, 2023-09-07:
Has been working fine so far with the correction of sunset here up ion the north in Finland
Today did not start.

Ubuntu 22.04.3. LTS

Here is what happened:
timo@Timo-TS10:~$ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/timo/.kivy/logs/kivy_23-09-07_3.txt
[INFO   ] [Kivy        ] v2.1.0
[INFO   ] [Kivy        ] Installed at "/home/timo/.local/lib/python3.10/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[INFO   ] [Config      ] Verifying station details
[INFO   ] [Factory     ] 189 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <sdl2>
[INFO   ] [GL          ] OpenGL version <b'4.6 (Compatibility Profile) Mesa 23.0.4-0ubuntu1~22.04.1'>
[INFO   ] [GL          ] OpenGL vendor <b'Intel'>
[INFO   ] [GL          ] OpenGL renderer <b'Mesa Intel(R) HD Graphics (CHV)'>
[INFO   ] [GL          ] OpenGL parsed version: 4, 6
[INFO   ] [GL          ] Shading version <b'4.60'>
[INFO   ] [GL          ] Texture max size <16384>
[INFO   ] [GL          ] Texture max units <32>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [GL          ] NPOT texture support is available
 Traceback (most recent call last):
   File "/home/timo/wfpiconsole/main.py", line 521, in <module>
     wfpiconsole().run()
   File "/home/timo/.local/lib/python3.10/site-packages/kivy/app.py", line 954, in run
     self._run_prepare()
   File "/home/timo/.local/lib/python3.10/site-packages/kivy/app.py", line 924, in _run_prepare
     root = self.build()
   File "/home/timo/wfpiconsole/main.py", line 199, in build
     self.screenManager.add_widget(CurrentConditions())
   File "/home/timo/wfpiconsole/main.py", line 440, in __init__
     self.app.astro.moonrise_moonset()
   File "/home/timo/wfpiconsole/lib/astronomical.py", line 198, in moonrise_moonset
     Moonset = self.observer.next_setting(self.moon)
   File "/home/timo/.local/lib/python3.10/site-packages/ephem/__init__.py", line 444, in next_setting
     return self._find_rise_or_set(body, start, use_center, +1, False)
   File "/home/timo/.local/lib/python3.10/site-packages/ephem/__init__.py", line 492, in _find_rise_or_set
     raise AlwaysUpError('%r is above the horizon at %s'
 ephem.AlwaysUpError: 'Moon' is above the horizon at 2023/9/8 17:23:06

peted-davis, 2023-09-10:
This is all part of the same bug that I am currently working on. The only fix at the moment is to disable the sunrise/sunset and moonrise/moonset functionality by commenting out lines 439, 440, 443 and 444 in `main.py`. I will get the fixed sorted ASAP!

TimoHav, 2023-09-10:
> Lähetetty: sunnuntai 10. syyskuuta 2023 15.29
> Vastaanottaja: peted-davis/WeatherFlow_PiConsole ***@***.***>
> Kopio: Timo Havulinna ***@***.***>; Author ***@***.***>

Hi Pete! Thank you for your quick answer!
I have two stations with two separate displays running your software. The first we made early this summer (sun was not setting in Finland) and did not start at all, but my son Antti fixed it with the patch that was available in Pull requests section.
The other one was made later this summer. So we will be waiting for your fix. I already bought some coffee to you and you can expect more then 😊

Best regards
Timo Havulinna
Finland
***@***.******@***.***>
Tel. +358 400 704704

TimoHav, 2023-09-11:
Could you also think of adding the current Battery Voltage somewhere on the screen? Below date and time? During the winter time we have to use the Tempest battery booster because there is not enough sun to keep the internal battery alive. Timo

pjmpessers, 2023-10-11:
@TimoHav 
Hello, Timo. This is possible, if you don't mind editing it yourself.
Go to directory /wfpiconsole/kvlang and open file rainfall.kv with a texteditor. Look for the line ## Time and date (I think #102) and place this:

## Time and date
    TimeDateField:
        text: '[color=ffffffff]' + app.CurrentConditions.System['Date'] + '[/color]'
        pos_hint: {'x': 0/262, 'y': 24/202}
        size_hint_x: (262/262)
    TimeDateField:
        text: '[color=ffffffff]' + app.CurrentConditions.System['Time'] + '[/color]'
        pos_hint: {'x': -45/262, 'y': 3/202}
        size_hint_x: (262/262)
    TimeDateField:                      
        text: '[color=ffffffff]' + app.CurrentConditions.Status['tempest_voltage']+"V" + '[/color]'
        pos_hint: {'x': 55/262, 'y': 3/202}
        size_hint_x: (262/262)

If i'm right it will work and you see smaething like this: 

[image]

TimoHav, 2023-10-12:
Hi! Thank you very much!

Best regards
Timo

Lähetetty: keskiviikko 11. lokakuuta 2023 23.57

Kopio: Timo Havulinna ***@***.***>; Mention ***@***.***>

[20231011_224118]<https://user-images.githubusercontent.com/34476404/274416063-7c162bc9-be69-4445-b7db-c231e2b815db.jpg>

pjmpessers, 2023-10-12:
Your welcome.

Verzonden vanaf mijn Galaxy

TimoHav, 2023-10-14:
And here with a picture of my display. Thank you again!

Best regards
Timo Havulinna
***@***.******@***.***>
Tel. +358 400 704704

Lähetetty: torstai 12. lokakuuta 2023 10.27

Vastaanottaja: peted-davis/WeatherFlow_PiConsole ***@***.***<mailto:***@***.***>>
Kopio: Timo Havulinna ***@***.***<mailto:***@***.***>>; Mention ***@***.***<mailto:***@***.***>>

pjmpessers, 2023-10-14:
@TimoHav I don't see anything. :-)

TimoHav, 2023-10-15:
Hi! It was there as an attachment. But sometimes email software won't let them through.
I will embed it in this email. Hope you see it now.

***@***.***

Best regards
Timo

Lähetetty: lauantai 14. lokakuuta 2023 18.04

pjmpessers, 2023-10-15:
Still not. :-)

TimoHav, 2023-10-15:
Very strange!!
How about through this link?
https://photos.app.goo.gl/spSdY5eA8FoVX4rb7

Best regards
Timo Havulinna
Tel +358 400 704704

Still not. :-)

TimoHav, 2023-10-27:
Hi Pete! I see that there is your update v23.10.1 available. You mentioned that in Facebook and now my devices also offer that on the screens.
What if I update will I lose our own corrections/additions like the Battery Voltage on the screen and the Sun/Moon according to the Pull Request section?

Best regards
Timo Havulinna
***@***.******@***.***>
Tel. +358 400 704704

peted-davis, 2023-10-27:
Hi Timo, yes I would hold off updating at the moment. I had hoped to roll out a big fix for the sunrise and sunset issues you are seeing in this release, but I haven't quite got there yet. It will be in the next release!

As for the extra voltage, yes that will be overwritten unless you have made a custom panel that is stored in the user/ directory. Custom panels in that directory are not overwritten when an update is released

TimoHav, 2023-10-27:
Ok thank you!

Best regards
Timo

Lähetetty: perjantai 27. lokakuuta 2023 15.34

TimoHav, 2023-10-28:
Hi Pete again!
Is there a way to eliminate the pop-up on the screen saying that there is an update available? After I put it away it comes again (once a day?).

Best regards
Timo Havulinna
***@***.******@***.***>
Tel. +358 400 704704

peted-davis, 2023-10-28:
> After I put it away it comes again (once a day?).

Yes, the console is scheduled to check for updates once a day. If you comment out line 206 of `main.py` (`Clock.schedule_once(self.system.check_version)`), you will never be prompted to update again

TimoHav, 2024-05-15:
Hi Pete!

Have you already included this in your recent updates?
I have halted the updates since October 2023

‘’Code to handle NeverUpError an AlwaysUpError, for Stations fare North and fare South’’

Best regards
Timo

Lähettäjä: Timo Havulinna
Lähetetty: perjantai 27. lokakuuta 2023 15.58
Vastaanottaja: peted-davis/WeatherFlow_PiConsole ***@***.***>
Aihe: VS: [peted-davis/WeatherFlow_PiConsole] Piconsole not starting (Issue #117)

Ok thank you!

Best regards
Timo

peted-davis, 2024-05-16:
Not yet - I'm sorry! I am close, but it still requires a bit more work. 

I know it sounds like it should be an easy problem to fix, but unfortunately it has turned into a much more complex problem. When I wrote this part of the code, I always assumed that the sun would rise above the horizon before it set beneath the horizon every day. This is not the case, however, at high northern and southern latitudes. When the night is very short, and in locations where solar noon is not aligned with local noon, the sun can set before it rises. Rewriting the console to handle (rather than fudge) these situations has/is taking some time. I will get there though!

TimoHav, 2024-05-17:
Hi Pete! No worries.
So far everything has worked fine at least in Finland for us. We did the corrections last autumn according to these:
https://github.com/peted-davis/WeatherFlow_PiConsole/pulls
We are in the southern parts of Finland.
Now up in the north 24 hour day has started and sun doesn’t set at all 😊

Best regards
Timo

Lähetetty: torstai 16. toukokuuta 2024 18.24

peted-davis, 2025-06-18:
Hi Timo, sorry for the delayed response. Did you forget to attach the logs? I don't see anything

peted-davis, 2025-06-18:
This should a relatively simple fix. Can you let me know which version you are running so I can get the line numbers right

TimoHav, 2025-06-18:
Or is it easy for you to list the lines which should be changed? Old line text -> corrected ?

Best regards
Timo

Lähettäjä: Timo Havulinna
Lähetetty: keskiviikko 18. kesäkuuta 2025 11.52
Vastaanottaja: peted-davis/WeatherFlow_PiConsole ***@***.***>; peted-davis/WeatherFlow_PiConsole ***@***.***>
Kopio: Mention ***@***.***>
Aihe: VS: [peted-davis/WeatherFlow_PiConsole] Piconsole not starting (Issue #117)

Hi Pete! Thank you for your answer. Unfortunately, I am now at our summer house and the unit is not running at my house. I will be back there beginning of July. I will let you know then.

Best regards
Timo

Lähetetty: keskiviikko 18. kesäkuuta 2025 11.30

peted-davis, 2025-06-23:
You should just be able to change line 274 in `sager.py` from `if METAR['clouds']:` to `if 'clouds' in METAR:`

If this doesn't fix the issue, we can dig a little deeper

TimoHav, 2025-06-23:
Thank you! I will try it when I get back home. Have a nice summer!

Best regards
Timo

Lähetetty: maanantai 23. kesäkuuta 2025 16.21

peted-davis, 2025-11-07:
Please run `wfpiconsole beta` followed by `wfpiconsole start`

If it still errors I will take a look into it 

Please don't run `wfpiconsole update` as that discards the beta version of the code

peted-davis, 2026-04-10:
You're not the first person to notice this rainfall issue today, although it doesn't seem to be affecting everyone. To find the cause I need to test a station that shows the issue. Would you mind sharing your `wfpiconsole.ini` file here? You can change your WeatherFlow Access Token afterwards to protect your data

Regarding the Sager Forecast, there still remains an issue with cloud information being provided by the API. Until the cloud data returns, the forecast will be marked as unavailable. This was the fix that was needed to stop the console crashing when the expected API field was not present

TimoHav, 2026-04-10:
Thank you for your quick comments.
I am happy to provide ini-files.
I have two units running your software. I will include both files.
If attachments don’t come properly you will find them also here:
https://drive.google.com/drive/folders/1X2QIeP8ZMWxpwDFC72ob0o4Lux0mw39c?usp=sharing

Best regards
Timo

Lähetetty: perjantai 10. huhtikuuta 2026 10.08

TimoHav, 2026-05-02:
Hi Pete! Thank you very much! Now it is up and running again. Date, Time and Voltage on the screen.

Best regards
Timo

Lähetetty: lauantai 2. toukokuuta 2026 19.10

Hi Timo, please change Date and Time in app.CurrentConditions.System['Date'] and app.CurrentConditions.System['Time'] to lower case: app.CurrentConditions.System['date'] and app.CurrentConditions.System['time']. Does that get you up and running again?