peted-davis/WeatherFlow_PiConsole #118: Issue with rain totals - different values than what appears on app

tw15egan, 2023-09-19:
Hey there! First off thanks for all your work on this app, I love it.

**Describe the bug**
I noticed yesterday that the totals in the display differ from what is displayed on my app.

**To Reproduce**
View screen

**Screenshots**
Display:
[image]

Yesterday
[image]

Month
[image]

Year
[image]

**Hardware/OS/Version (please complete the following information):**
 - OS: Debian Buster / Bullseye (two devices)
 - Hardware: pi3 + pi4
 - Console Version: Latest, just tried updating

**Additional context**
The only thing I can think of is that I had turned off the "nearcast data" for a few days to see if I got wildly different readings, but I ended up turning it back on. I did this about 2 weeks ago and turned it back on within a few days (All would have been within the month of September, but year totals are still skewed). Is there any way to "refresh" the data on the display with what is stored on in the app?

Thanks again for your help!

peted-davis, 2023-09-30:
This will be a NearCast rain issue. The console does not display NearCast values, so if you have NearCast enabled in the app, then the rain values will differ compared to the console

tw15egan, 2023-10-02:
Excellent, thanks for the clarification 👍🏻

closed 2023-10-02 by tw15egan