peted-davis/WeatherFlow_PiConsole #120: WFpiconsole can't be installed on debian bookworm 32 bits.

pjmpessers, 2023-10-11:
**Describe the bug**
When using curl -sSL https://peted-davis.github.io/wfpiconsole | bash got a error

See screenshot.
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: Debian bookworm 32 bits
 - Hardware: PI 4B

peted-davis, 2023-10-12:
Yes, this PiConsole is not ready yet for Debian Bookworm. I am working on the necessary changes, however.

pjmpessers, 2023-10-12:
Great to hear. I ordered a raspberry 5, because i installed wfpiconsole, sharptools dashboard and home assistant and it is a bit sluggish. 😊

Verzonden vanaf mijn Galaxy

pjmpessers, 2023-10-20:
How can i install wfpiconsole from the develop branch? curl -sSL https://peted-davis.github.io/wfpiconsole | bash will install the main branch.

peted-davis, 2023-10-26:
This is now fixed in v23.10.1. There is no official support yet for Raspberry Pi 5, but it will come

closed 2023-10-26 by peted-davis

pjmpessers, 2023-10-27:
Thank you. It works under debian bookworm 32 bits with RPI 4b 8gb using X11.