peted-davis/WeatherFlow_PiConsole #123: Python error after upgrading to v23.10.1

cmoo92, 2023-10-26:
**Describe the bug**
After running the upgrade, when I tried to start the console I got this error message:

`/usr/local/bin/wfpiconsole: line 811: /home/username/wfpiconsole/venv/bin/python3: No such file or directory
`
I was able to fix the issue by running the `wfpiconsole install` command again.

**Hardware/OS/Version (please complete the following information):**
 - OS: Raspberry Pi OS Bullseye
 - Hardware: Pi 4 Model B
 - Console Version: 23.10.1

peted-davis, 2023-10-26:
Thanks for the bug report - I'll take a look. Can you open a terminal and run this command: `echo $USER` and let me know the output?

cmoo92, 2023-10-26:
There is no output when I run that command

peted-davis, 2023-10-26:
> The only output is just my username

What is your username? According to the error you shared, the PiConsole thinks your username is 'username'. Is that correct?

cmoo92, 2023-10-26:
Oh, sorry for the confusion. I had just modified it in my original submission to obfuscate my actual username and replaced it with the generic term "username". The` echo $USER` command does return my actual username.

peted-davis, 2023-10-26:
Ah I see! So to double check, `echo $USER` returns the correct username, and that also matches the username that was shown in the error before you obfuscated? If so, it's not an error in the path, and I'll have to keep investigating

cmoo92, 2023-10-26:
That is correct, user names were the same so it's not an error in the path.

peted-davis, 2023-10-26:
How exactly did you run the upgrade? Did you do `wfpiconsole update` or did you call the update function in the `wfpiconsole.sh` file directly?

peted-davis, 2023-10-26:
OK, this should be fixed now. There was an issue with the installer where it was deleting the new `venv` folder while fetching the latest version of the code.

closed 2023-10-26 by peted-davis