peted-davis/WeatherFlow_PiConsole #124: Tried to update because I’m being told on screen there is a new version

donaldnaegele, 2023-10-26:
First updated to the latest raspberry pi. No issues with that update 

then tried to update with the wpficonsole update command. 

It was working then stopped with errors. 

I rebooted and started the process again. 

I’m getting an error message. 

Error: unable to update local package cache. Please check your internet connection. 

Reading package lists…

E: could not get lock var/lib/apt/lists/lock. It is held by process 1449 (packa      Then the screen cuts off. Then says getitd)

E: unable to lock directory /var/lib/apt/lists 

The app still works but tells me I need to update still. 

I don’t know how to tell what pi or version I have. 
Help?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: [e.g. Raspberry Pi OS Bullseye]
 - Hardware: [e.g. Pi3]
 - Console Version: [e.g. 22.3.1]

**Additional context**
Add any other context about the problem here.

peted-davis, 2023-10-26:
> First updated to the latest raspberry pi. No issues with that update

What do you mean by this? What commands did you run?

When you run the `wfpiconsole update` command, can you copy and paste here everything that is printed to the terminal please?

donaldnaegele, 2023-10-26:
I ran Sudo apt update. Then I ran sudo apt full-upgrade. Then I rebooted. Then I ran wfpiconsole update. Then it started. Processed for a while. Tons of errors. I didn’t screen capture those. I then ran sudo reboot now. Then I ran wfpiconsole update and got the error above.

peted-davis, 2023-10-26:
OK, so the error you are seeing is to do with your systems package manager (apt). It is not an error with the PiConsole per se, but the update won't install while there is an issue with the package manager (because the console needs to be able to install packages). At this point it might be easier to start over with a new install of Raspberry Pi OS to fix the bigger package manager  problem

donaldnaegele, 2023-10-26:
I did a little Google and found this command. 

sudo apt-get dist-upgrade

Then I tried the wfpiconsole update and it worked. Rebooted. No longer getting the update needed message on the bottom right. 

Don.

peted-davis, 2023-10-27:
Great! Glad you found the fix

closed 2023-10-27 by peted-davis