peted-davis/WeatherFlow_PiConsole #125: Console start but not shown screen

zuper83, 2023-10-27:
Hi,

Thanks for and outstanding work and dedication to this project and helping us fellow users! Highly appreciated!

Scenario
Ran Pi3 Buster but updated to Bullseye as part of breaking changes.
Ran wfpiconsole update and everything went smooth.
Running wfpiconsole start and got this far but nothing turns up on the screen as it used to to.

[image]

Only thing that stands out in my pi config is lcdrotate=2. Commented it out without any change to the startup process. I'm able to ping my hub so i guess nothing wrong with the network.
Using a standard 7" screen and everything worked smooth with WFpiconsole prior Bullseye and the latest update :-)

Any suggestions?

Thanks in advance.

jsb2092, 2023-10-27:
Same issue here.  It says it started, but nothing ever shows on screen.

peted-davis, 2023-10-27:
Sorry you're both having issues. Can you confirm how you updated to Bullseye? Did you flash a fresh image to your SD card, or did you try an in-place upgrade. 

Are you using 32 bit or 64 bit?

jsb2092, 2023-10-27:
I did an in place upgrade, following these instructions: https://www.tomshardware.com/how-to/upgrade-raspberry-pi-os-to-bullseye-from-buster.  Running on  Raspberry Pi 3 Model B Plus Rev 1.3.

Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

zuper83, 2023-10-27:
32 bit. I also did a place in upgrade using same guide as @jsb2092

[image]

peted-davis, 2023-10-28:
Ok, I suspect the in place upgrade is the issue. Raspberry Pi OS is known to not always upgrade smoothly using this method. This snippet is from the guide you followed:

> The ideal way to install the latest OS is to backup all your project and data files and then do a clean install using Raspberry Pi Imager to write to a microSD card, but it is possible to upgrade an old Debian 10 “Buster” based Raspberry Pi OS to Bullseye.There is a caveat, however, as it may not be the smoothest of upgrades.

I would suggest downloading a fresh copy of Bullseye and flash that to your SD card instead

zuper83, 2023-10-28:
Thanks for your investigation and reply. 

Did a fresh install as per suggestion. Rasbian Bullseye (Lite) 32bit. Installed wfpiconsole. Installation got an "error" and mentioned pthyon3-venv needed and aborted installation. Installed pthyon3-venv. Run installation wfpiconsole again and it finalized without issues this time. Chose option 2 - UDP and Rest-API. Then...

pi@wfpiconsole:~ $ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_23-10-28_3.txt
[INFO   ] [Kivy        ] v2.2.0
[INFO   ] [Kivy        ] Installed at "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110]
[INFO   ] [Python      ] Interpreter at "/home/pi/wfpiconsole/venv/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[INFO   ] [Config      ] Verifying station details
[INFO   ] [Factory     ] 190 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2 (img_pil, img_ffpyplayer ignored)
[ERROR  ] [Input       ] MTDev is not supported by your version of linux
Traceback (most recent call last):
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/input/providers/__init__.py", line 41, in <module>
    import kivy.input.providers.mtdev
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/input/providers/mtdev.py", line 94, in <module>
    from kivy.lib.mtdev import Device, \
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/lib/mtdev.py", line 30, in <module>
    libmtdev = cdll.LoadLibrary('libmtdev.so.1')
  File "/usr/lib/python3.9/ctypes/__init__.py", line 452, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmtdev.so.1: cannot open shared object file: No such file or directory
[INFO   ] [Window      ] Provider: sdl2(['window_egl_rpi'] ignored)
[CRITICAL] [Window      ] Unable to find any valuable Window provider. Please enable debug logging (e.g. add -d if running from the command line, or change the log level in the config) and re-run your app to identify potential causes
egl_rpi - ImportError: cannot import name 'bcm' from 'kivy.lib.vidcore_lite' (/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/lib/vidcore_lite/__init__.py)
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/core/__init__.py", line 59, in core_select_lib
    mod = importlib.__import__(name='{2}.{0}.{1}'.format(
  File "<frozen importlib._bootstrap>", line 1109, in __import__
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/core/window/window_egl_rpi.py", line 12, in <module>
    from kivy.lib.vidcore_lite import bcm, egl

sdl2 - RuntimeError: b'Could not initialize OpenGL / GLES library'
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/core/__init__.py", line 71, in core_select_lib
    cls = cls()
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/core/window/window_sdl2.py", line 165, in __init__
    super(WindowSDL, self).__init__()
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/core/window/__init__.py", line 1102, in __init__
    self.create_window()
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/core/window/window_sdl2.py", line 316, in create_window
    self.system_size = self._win.setup_window(
  File "kivy/core/window/_window_sdl2.pyx", line 235, in kivy.core.window._window_sdl2._WindowSDL2Storage.setup_window
  File "kivy/core/window/_window_sdl2.pyx", line 76, in kivy.core.window._window_sdl2._WindowSDL2Storage.die

x11 - ImportError: libXrender.so.1: cannot open shared object file: No such file or directory
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/core/__init__.py", line 59, in core_select_lib
    mod = importlib.__import__(name='{2}.{0}.{1}'.format(
  File "<frozen importlib._bootstrap>", line 1109, in __import__
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 565, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1108, in create_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed

[INFO   ] [Text        ] Provider: sdl2
Traceback (most recent call last):
  File "/home/pi/wfpiconsole/main.py", line 596, in <module>
    wfpiconsole_app.run()
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/app.py", line 955, in run
    self._run_prepare()
  File "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/app.py", line 925, in _run_prepare
    root = self.build()
  File "/home/pi/wfpiconsole/main.py", line 186, in build
    self.set_scale_factor(self.window, self.window.width, self.window.height)
AttributeError: 'NoneType' object has no attribute 'width'

peted-davis, 2023-10-28:
You can't use the Lite version. It doesn't come with a desktop so it has no way of displaying graphical applications. You need the full version, with desktop, but without recommended software (unless you want this)

jsb2092, 2023-10-28:
> Ok, I suspect the in place upgrade is the issue. Raspberry Pi OS is known to not always upgrade smoothly using this method. This snippet is from the guide you followed:
> 
> > The ideal way to install the latest OS is to backup all your project and data files and then do a clean install using Raspberry Pi Imager to write to a microSD card, but it is possible to upgrade an old Debian 10 “Buster” based Raspberry Pi OS to Bullseye.There is a caveat, however, as it may not be the smoothest of upgrades.
> 
> I would suggest downloading a fresh copy of Bullseye and flash that to your SD card. I tested this multiple times before releasing and it worked fine

While I did read that I assumed that the upgrade process would fail, as it did for a friend of mine. I assumed if the upgrade actually completed then things would work.  While I am not opposed to doing this, it is a hassle, and given that the upgrade notice makes the console unusable I'd prefer that there would be a way to ignore upgrades if the process will continue to be this invasive.

peted-davis, 2023-10-28:
> I'd prefer that there would be a way to ignore upgrades

Sure, if you comment out line 206 of `main.py` (`Clock.schedule_once(self.system.check_version)`), you will never be prompted to update again

jtg7330, 2023-10-28:
Hi,  I also have the same issue on a clean install of Bullseye this morning
[image]

jsb2092, 2023-10-28:
> > I'd prefer that there would be a way to ignore upgrades
> 
> 
> 
> Sure, if you comment out line 206 of `main.py` (`Clock.schedule_once(self.system.check_version)`), you will never be prompted to update again
> 
> 

Fair.  Though I was more referring to getting a notice but being able to clear it from the ui.

zuper83, 2023-10-29:
I'm up running again. Thanks for your support.

peted-davis, 2023-10-29:
> Fair. Though I was more referring to getting a notice but being able to clear it from the ui

Unfortunately there is no way to just get the notice once. You either get it once a day, or not at all if you comment out that line. I am going to try and make the notification more user friendly, but that will require you to update at some point

peted-davis, 2023-10-29:
> Hi, I also have the same issue on a clean install of Bullseye this morning. Using Pi 3b on 32 bit bullseye 5-03-2023 version

Are you trying to start the console over an SSH connection, or are you starting it directly from the terminal on the Pi? I have retested this exact setup just now and I have no problems with it. I don't see `Unable to connect to X server` which is the root cause of your problem

jsb2092, 2023-10-29:
I did a reinstall, and it looks like the problem is that the upgrade doesn't start the desktop.

peted-davis, 2023-10-29:
> I did a reinstall, and it looks like the problem is that the upgrade doesn't start the desktop.

Ah I see. So you are back up and running now?

jsb2092, 2023-10-29:
yes

jtg7330, 2023-10-29:
> Are you trying to start the console over an SSH connection, or are you starting it directly from the terminal on the Pi? I have retested this exact setup just now and I have no problems with it. I [...]

That was it exactly, using the SSH console. All resolved and running fine. much thanks!

closed 2023-11-06 by zuper83