peted-davis/WeatherFlow_PiConsole #127: wfpiconsolewfpiconsole.log in ~/

MABeatty1978, 2023-10-30:
**Describe the bug**
The log file is being written to ~/ and has the name wfpiconsolewfpiconsole.log

I'm guessing this the result of a bug in wfpiconsole.service as the StandardOutput and StandardError variables are missing the "/" in what is supposed to be wfpiconsole/wfpiconsole.log

**Screenshots**

[image]

peted-davis, 2023-10-30:
Thanks for spotting this. I can confirm this is a bug and a / is missing. I'll get this fixed up

peted-davis, 2023-11-06:
This should now be fixed in v23.11.1

closed 2023-11-06 by peted-davis