peted-davis/WeatherFlow_PiConsole #128: Thermometer icon showing up

MABeatty1978, 2023-10-30:
**Describe the bug**
This seems to be random and I cannot figure out how to replicate it, it just shows up for a while and then goes away.  See circled icon in the screenshot

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add 
[image]
screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: Bookworm
 - Hardware: Pi3
 - Console Version: Most recent upgrade

peted-davis, 2023-10-30:
That's a Raspberry Pi thing showing that your Pi is getting too hot and is overheating. The OS will be throttling the CPU to keep the temperatures in check. You'll probably want to consider improving the airflow to the Pi

MABeatty1978, 2023-10-30:
Ooooffff.... I knew that too. Sorry

peted-davis, 2023-10-30:
No problem! It is confusing when seen on a weather display 😂

closed 2023-10-30 by peted-davis