peted-davis/WeatherFlow_PiConsole #129: REST-api Warning?

zuper83, 2023-10-31:
Hi,

Maybe not an issue, more to understand the basics. 

My current settings

[System]
Connection = UDP
rest_api = 1
SagerInterval = 6
Timeout = 20
Hardware = Pi3
Version = v23.10.1

For two days now my log is flodded every minute with.

[[1;33mWARNING[0m] [request_api ] 2023-10-31 10:04:23 - last_24h call failed
[[1;33mWARNING[0m] [request_api ] 2023-10-31 10:04:23 - Today call failed
[[1;33mWARNING[0m] [request_api ] 2023-10-31 10:04:24 - Yesterday call failed
[[1;33mWARNING[0m] [request_api ] 2023-10-31 10:04:24 - Month call failed
[[1;33mWARNING[0m] [request_api ] 2023-10-31 10:04:25 - Year call failed

Nothing accumulates on the screen. I have restarted PI console and and awaited at least 24h but nothing changes same log warnings, repeates. I did even restart the hub as desperate response and awaited at least 24h. Both API keys enetered thru the desktop. Sager and forecast works just fine with above settings. If i switch to websocket everything works just fine, nor warnings. But i'd like to go for UDP and REST-API (history related if I understand correct)

Are those warning normal or an issue?

peted-davis, 2023-10-31:
Thanks for the report. This is definitely a bug. With `rest_api = 1` you should see fields accumulate and get the forecast data. I will take a look to see what is going on. Sorry I didn't spot this before the release

zuper83, 2023-11-01:
Ok. Thanks for the clarification. Then it will be websocket connection until further noticed.

peted-davis, 2023-11-05:
This should now be fixed in v23.11.1

zuper83, 2023-11-06:
Great. Log looks fine after update. Let´s close this issue.

closed 2023-11-06 by zuper83