peted-davis/WeatherFlow_PiConsole #130: Cannot get past configuration

bgiuriceo, 2023-11-01:
**Describe the bug**
Installation hangs at
 Do you own an Indoor AIR?* [y/n] n
  Adding height of TEMPEST: 4.572
Traceback (most recent call last):
  File "/home/pi/wfpiconsole/main.py", line 42, in <module>
    configFile.create()
  File "/home/pi/wfpiconsole/lib/config.py", line 198, in create
    write_config_key(config, section, key, default[section][key])
  File "/home/pi/wfpiconsole/lib/config.py", line 687, in write_config_key
    value = STATION['stations'][idx][key.lower()]
**TypeError: list indices must be integers or slices, not NoneType**
pi@wfpiconsole:~ $

**Hardware/OS/Version (please complete the following information):**
 - OS: [e.g. Raspberry Pi OS Bullseye or Bookworm
 - Hardware: Pi3B+ 
 - Console Version: ???

**Additional context**
This started when I needed to update from Buster so I did a clean install.

bgiuriceo, 2023-11-01:
Sorry I left off that I tried to do a fresh install on Bookworm and Bullseye on gear that was working beautifully for 2 years, I think.

peted-davis, 2023-11-01:
Sorry you've run into trouble. There is some peculiarity in your station info that is causing the console to crash. The station info is only needed to configure the console, which is why this has only cropped up now. 

To investigate further, I ideally need to know your station ID, Tempest device ID, and your personal access token so I can get your station info from the API. Once the bug is fixed, you can delete the access token and create a new one to ensure no one else has access to your data. If you are a member community.weatherflow.com you can send me a private message there with the details if you'd prefer

bgiuriceo, 2023-11-01:
HI Pete Thanks for the quick reply

WeatherFlow Access Token*: 041fee4d-eb17-4d2b-9fd7-28f5d905f037
CheckWX API Key*: dc1701186ae5460dabbc237903

Station ID*: 00059011
TEMPEST device ID*: 92428
TEMPEST serial number: ST-00059011

Thanks again
b

> Sorry you've run into trouble. There is some peculiarity in your station info that is causing the console to crash. The station info is only needed to configure the console, which is why this has [...]

Bruce Giuriceo
639 Harmony Way
Centreville, MD 21617-2489 USA

mobile  (443) 885-0885
fax  (443) 4bg-fax4

Effective September 1, 2023 my email will become:
***@***.***

bgiuriceo, 2023-11-01:
I’m a fool! Wow sorry, I used the wrong station ID…it should be:
Station ID*: 28774
So that error is now corrected. Config continues until I get:

Unable to connect to X server

***@***.***:~ $ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_23-11-01_4.txt
[INFO   ] [Kivy        ] v2.2.0
[INFO   ] [Kivy        ] Installed at "/home/pi/wfpiconsole/venv/lib/python3.9/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110]
[INFO   ] [Python      ] Interpreter at "/home/pi/wfpiconsole/venv/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[INFO   ] [Config      ] Verifying station details
[INFO   ] [Factory     ] 190 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: sdl2(['window_egl_rpi'] ignored)
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <sdl2>
[INFO   ] [GL          ] OpenGL version <b'2.1 Mesa 20.3.5'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'VC4 V3D 2.1'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 1
[INFO   ] [GL          ] Shading version <b'1.20'>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <16>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [GL          ] NPOT texture support is available
Unable to connect to X server
[INFO   ] [Base        ] Start application main loop
[INFO   ] [Websocket   ] 2023-11-01 18:12:00 - Opening connection
[INFO   ] [Websocket   ] 2023-11-01 18:12:01 - Connection open

> 
> HI Pete Thanks for the quick reply
> 
> WeatherFlow Access Token*: 041fee4d-eb17-4d2b-9fd7-28f5d905f037
> CheckWX API Key*: dc1701186ae5460dabbc237903
> 
> Station ID*: 00059011
> TEMPEST device ID*: 92428
> TEMPEST serial number: ST-00059011
> 
> Thanks again
> b
> 
> 
> 
>> 
>> 
>> Sorry you've run into trouble. There is some peculiarity in your station info that is causing the console to crash. The station info is only needed to configure the console, which is why this has only cropped up now.
>> To investigate further, I ideally need to know your station ID, Tempest device ID, and your personal access token so I can get your station info from the API. Once the bug is fixed, you can delete the access token and create a new one to ensure no one else has access to your data. If you are a member community.weatherflow.com you can send me a private message there with the details if you'd prefer
> 
> Bruce Giuriceo
> 639 Harmony Way
> Centreville, MD 21617-2489 USA
> 
> mobile  (443) 885-0885
> fax  (443) 4bg-fax4
> 
> Effective September 1, 2023 my email will become:

Bruce Giuriceo
639 Harmony Way
Centreville, MD 21617-2489 USA

mobile  (443) 885-0885
fax  (443) 4bg-fax4

Effective September 1, 2023 my email will become:
***@***.***

bgiuriceo, 2023-11-01:
After invoking auto start the app works fine. There is some refesh issue on the clock where the display is breaking up. Maybe I'll try a Bookworm install now. 
Thanks so much for great support and a program that I learned that i can't live without....
cheers!

closed 2023-11-01 by bgiuriceo

peted-davis, 2023-11-02:
Glad you managed to get this fixed. The code should have alerted you to the incorrect station ID rather than just crash out, but there was bug preventing this from happening. I have fixed this bug for the future. Thanks for the report