peted-davis/WeatherFlow_PiConsole #131: Numpy - error when updating to newest version of PiConsole

DMEbner, 2023-11-04:
Hello friends!

I have been trying to the most current version of Pi Console but I keep getting error that it can't install the numpy module. 
[image]

Numpy is loaded on my Rasp. Pi. Can anyone give me some direction? 

Cheers!

peted-davis, 2023-11-04:
Sorry you have run into trouble. It looks like this happens if Numpy is already installed on the Pi. To fix the issue, please open a terminal and run the following commands:

`source ~/wfpiconsole/venv/bin/activate`
`python3 -m pip install --no-cache-dir numpy==1.26.0`
`deactivate`

Once those three commands have run, try to update the console again with `wfpiconsole update`

DMEbner, 2023-11-05:
Thank you for the help. It says there is no such file or directory. Should
I just reinstall Pi Console on my Pi?

peted-davis, 2023-11-05:
Yes, just try reinstalling. I have just pushed an update that should fix the issue you are seeing

peted-davis, 2023-11-06:
This error should now be fixed in v23.11.1

closed 2023-11-06 by peted-davis