peted-davis/WeatherFlow_PiConsole #133: Display offset from top on Raspberry Pi 7" touchscreen

pditterline, 2023-11-06:
Since upgrading to 23.10.1, the console's display has been shifted down a bit on my touchscreen (I would guess ~60-70 pixels), hiding the bottom of the console. I recently updated my OS packages as well so that could be to blame, but the problem didn't start until the console update. The console works fine, the display is just off. I updated to 23.11.1 this morning but the issue persists.

Repro steps:
Reproducing my entire environment would be tough, though it's pretty much a stock rpi4 running bullseye with the official 7" touchscreen. I'm running the console via a desktop shortcut:
```
[Desktop Entry]
Name=WeatherFlow PiConsole
Comment=Opens the WeatherFlow console
Icon=/usr/share/pixmaps/openbox.xpm
Exec=/usr/local/bin/wfpiconsole start
Type=Application
Encoding=UTF-8
```

Screenshot from VNC:
[image]

**Hardware/OS/Version (please complete the following information):**
 - OS: Raspberry Pi OS Bullseye
 - Hardware: Pi4
 - Console Version: 23.11.1

**Additional context**
I haven't been able to use the touchscreen to control the console for quite a while, though it initially worked when I set things up. Probably irrelevant, but just in case. :)

I don't see any errors in the wfpiconsole.log, and the final entries look like they're reading the resolution correctly:
```
[INFO   ] [HIDMotionEvent] using <raspberrypi-ts^@>
[INFO   ] [HIDMotionEvent] <raspberrypi-ts^@> range ABS X position is 0 - 799
[INFO   ] [HIDMotionEvent] <raspberrypi-ts^@> range ABS Y position is 0 - 479
[INFO   ] [HIDMotionEvent] <raspberrypi-ts^@> range position X is 0 - 799
[INFO   ] [HIDMotionEvent] <raspberrypi-ts^@> range position Y is 0 - 479
```

peted-davis, 2023-11-09:
I'm not entirely sure what is going on here, sorry! This isn't a persistent bug that I can recreate, so I suspect it is something very specific with your system. If it isn't too onerous, I would suggest starting again with a fresh install of Bullseye and see if that fixes the issue.

pditterline, 2023-11-09:
I figured that might be the case. :) I'll see what I can find and report back if I can fix it in case others have similar issues.

pditterline, 2023-11-09:
Hmm.. found it, sort of. The default height in `.kivy/config.ini` is 600, and even though `wfpiconsole.ini` has a (correct) height of 480, it's not writing that to the `.kivy/config_wfpiconsole.ini` file when it starts up. Looking through the code in `main.py`, it looks like it should be doing that correctly, but it's not for some reason.
https://github.com/peted-davis/WeatherFlow_PiConsole/blob/4b703912d60da0e27843e8fe2b007bb92a1f4198/main.py#L84

I manually changed the height value in config.ini to 480 and restarted the console, and it looks correct now. The `config_wfpiconsole.ini` is being rewritten on startup so it doesn't look like a permission issue.

peted-davis, 2023-11-10:
OK, glad you managed to find a fix, but I'm still not entirely sure what is going on here. I have tried changing the width, height and full screen settings on my Pi4 test rig, and everything is working as expected. I think we'll close this for now, unless it starts cropping up more widely as an issue

closed 2023-11-10 by peted-davis