peted-davis/WeatherFlow_PiConsole #136: Time display is jerky.

geewizard1, 2023-11-24:
The latest version is making the time updates no longer smoothly incrementing.  The seconds digits jump from one value to several seconds ahead.  And that increment is variable.
Is this just the way it is?

WFPiConsole version is the newest.
RPi OS is Bullseye.
Rpi is a 3B.

peted-davis, 2024-02-09:
I've been away for a while so haven't been able to reply. Do you still see this issue?

geewizard1, 2024-02-09:
Yes, it is still occuring.

Bob

peted-davis, 2024-02-10:
Ok I suspect to a degree this is just the way it is now. A Pi3 is old hardware compared to the Pi4 and Pi5, and many of the underlying graphics modules are optimised to work on the newer hardware (something that is out of my control), which often comes at cost when running on older hardware

geewizard1, 2024-02-11:
No worries.  Thanks for getting back to me.

closed 2024-02-11 by peted-davis