peted-davis/WeatherFlow_PiConsole #138: console periodic reboots

TJ51FIFTY, 2024-01-31:
**Describe the bug**
A clear and concise description of what the bug is.

First, thank you for making this.  I've owned a Tempest for over 3 years and just set up PiConsole with a spare 7 inch LCD touchscreen and a Pi 3.   Added an acrylic frame and it is wonderful.

it is set to auto boot the console.  periodically I see the console crash to desktop then immediately reload.   it seems infrequent (every hour or so) but when it restarts it may do it 2 or 3 times.

this is a fresh OS install, and everything on the pi seems to function as it should.  I'm using the official power supply.

any ideas what would cause this?  I've searched here for similar symptoms but didn't see anything.

this happens so infrequently I can live with it, but I'd like to figure it out.

thank you.   

**Hardware/OS/Version (please complete the following information):**
 - OS: Linux raspberrypi 6.1.21-v7+
 - Hardware: Pi 3 b v1.2
 - Console Version: 23.11.1

**Additional context**
Add any other context about the problem here.

tw15egan, 2024-02-14:
@TJ51FIFTY Did you ever figure this out? Having the same issue.

TJ51FIFTY, 2024-02-14:
No, still random periodic console restarts on two pi console displays.   one a pi3b, one a pi 4 (never noticed them rebstarting at the same time).   very odd.

peted-davis, 2024-02-14:
I have been away so didn't look into this before the issue was closed. To look into this further I need to see the log files when the console crashes. The log file is located in the `wfpiconsole` directory `~/wfpiconsole/wfpiconsole.log`. The log file is overwritten each time the console runs so it will need to be captured before the console restarts

closed 2024-04-30 by peted-davis

peted-davis, 2025-08-16:
This is strange - I haven't heard or seen this before. Glad it's working though!