peted-davis/WeatherFlow_PiConsole #139: Sager Forecast Unavailable

swvalenti194, 2024-02-07:
Thanks for this great module.  Recently the sager forecast hasn't been displaying.  Checked my wx account and api is checking in as it should from the console.  Any ideas?

swvalenti194, 2024-02-09:
Was told by checkwx that the python script is the issue

peted-davis, 2024-02-09:
What exact error are you seeing on the Sager Forecast panel?

swvalenti194, 2024-02-10:
@peted-davis says "forecast unavailable"

peted-davis, 2024-02-10:
Ok, what is your Station ID?

swvalenti194, 2024-02-10:
132344

swvalenti194, 2024-02-14:
@peted-davis Any update or anything i can do to assist?

peted-davis, 2024-02-14:
I haven't been able to recreate this issue on my test setup. On the CheckWX API website can you revoke your API key and generate a new one and see if that works. If not, can you share the new API key with me as well as the latitude and longitude of your station? That will enable me to test as if I was using your station. You can revoke and regenerate the API key once I have finished with it to prevent others from using it

swvalenti194, 2024-02-16:
Ok great will report bank thank you!

swvalenti194, 2024-02-18:
@peted-davis changed API to no avail, PM you the details.

peted-davis, 2024-02-21:
@swvalenti194, ok I have checked out your key and lat/lon and I don't see any issues at my end. This suggests that the issue is either to do with the data being provided by your Tempest (which is also required for the Sager Forecast), or you are having network issues that is preventing the CheckWX API call succeeding. Ideally I need your complete `wfpiconsole.ini` file to move fowards as this will allow me to investigate the data coming from your Tempest device. You can either share it here (but it will be public), or if you are a member of the Tempest forum (community.weatherflow.com) you can send me a private message. My username over there is `peter`

swvalenti194, 2024-02-21:
@peted-davis I will send it over, it worked for a couple of weeks when I setup the unit but suddenly stopped.

peted-davis, 2024-02-23:
You should be able to click on my profile picture next to any post I have created and select the blue `Message` button. If you have only just joined that forum your 'level' might be too low to send messages. What is your username and I will PM you?

swvalenti194, 2024-02-23:
@peted-davis swvalenti

closed 2024-03-31 by peted-davis