peted-davis/WeatherFlow_PiConsole #140: Documentation Note - Change of Weatherflow Station and TempestSN

reubenfarrelly, 2024-03-01:
This is just a request for a minor update to the documentation - on the basis of an experience I have just had.

Recently my original Weatherflow station started to suffer from low battery charge and was powering off overnight despite being in full sun in summer, and I contacted support to resolve.  They sent me a whole new station which I installed a few days ago.

Since then however, all statistics and numbers in the PiConsole stopped updating.  The web view was fine but nothing was updating in the console.  I am set to use UDP broadcast for live data.  I did a "replace device" in the app.

It turns out that if the station serial number is changed, then a corresponding change needs to be made in wfpiconsole.ini to the TempestSN value.  If not, you'll get no data.

Mine is now fixed and working fine, but I think it's worth a one line comment in the instructions that if the station is changed, then wfpiconsole.ini also needs to be updated.  Or ideally dynamically learn of the TempestSN from the hub itself.

peted-davis, 2024-03-01:
Thanks for the comment, I will update the documentation accordingly. 

As it is possible for a hub to host multiple Tempest devices, the console cannot automatically determine from which device data should be displayed. Equally, when a new device appears on a hub, the console cannot determine if this is a replacement, or an addition of a new device. Hence input is required from the user

peted-davis, 2024-05-29:
This has now been added to the [README](https://github.com/peted-davis/WeatherFlow_PiConsole?tab=readme-ov-file#advanced-device-replacement)

closed 2024-05-29 by peted-davis