peted-davis/WeatherFlow_PiConsole #141: Constant relaunch

swvalenti194, 2024-03-25:
Just recently I am getting constant restarts of wfpiconsole even though the connection is never lost.  Keeps quitting and relaunching.  Any ideas?

{PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye}

swvalenti194, 2024-03-25:
Here is the log before a crash

[INFO   ] [Websocket   ] 2024-03-23 18:55:38 - Opening connection
[ERROR  ] [Websocket   ] 2024-03-23 18:55:48 - General error:
[INFO   ] [Websocket   ] 2024-03-23 18:55:58 - Opening connection
[ERROR  ] [Websocket   ] 2024-03-23 18:56:08 - General error:
[INFO   ] [Websocket   ] 2024-03-23 18:56:18 - Opening connection
[ERROR  ] [Websocket   ] 2024-03-23 18:56:28 - General error:
[INFO   ] [Websocket   ] 2024-03-23 18:56:38 - Opening connection
[INFO   ] [Websocket   ] 2024-03-23 18:56:43 - Connection open
[WARNING] [request_api ] 2024-03-23 18:57:03 - last_24h call failed
[WARNING] [Websocket   ] 2024-03-23 18:57:26 - Ping failed
[INFO   ] [Websocket   ] 2024-03-23 18:57:26 - Closing connection
[INFO   ] [Websocket   ] 2024-03-23 18:57:26 - Connection closed
[INFO   ] [Websocket   ] 2024-03-23 18:57:36 - Opening connection
[INFO   ] [Websocket   ] 2024-03-23 18:57:36 - Connection open
[WARNING] [Websocket   ] 2024-03-23 18:58:36 - Ping failed
[INFO   ] [Websocket   ] 2024-03-23 18:58:36 - Closing connection
[INFO   ] [Websocket   ] 2024-03-23 18:58:36 - Connection closed
[INFO   ] [Websocket   ] 2024-03-23 18:58:46 - Opening connection
[INFO   ] [Websocket   ] 2024-03-23 18:58:46 - Connection open

peted-davis, 2024-03-31:
This doesn't show the console crashing - it just shows that the websocket connection is resetting constantly. Likely a network error

swvalenti194, 2024-03-31:
Interesting thank you. Not sure why but I'll see if I can drill that down.

swvalenti194, 2024-04-03:
@peted-davis Not sure why this is happening now, didn't happen for months.  Nothing in network has changed.  Any ideas to see what is happening?

closed 2024-04-20 by swvalenti194