peted-davis/WeatherFlow_PiConsole #142: WFPICONSOLE result in a crash.

pjmpessers, 2024-05-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to WFPICONSOLE updat in terminal.
2. 
4. See error 
[â] OS check passed (Raspbian GNU/Linux 12 (bookworm))
  [!] Raspberry Pi check warning (Raspberry Pi 5 Model B Rev 1.0)

  [!] WARNING: unsupported architecture or Raspberry Pi detected
      No support is available for errors encountered while running
      the PiConsole

  ============================== 
  Updating WeatherFlow PiConsole 
  ============================== 

  [â] Checking for updated packages
  Error: Unable to update local package cache. Please check your internet connection

 Ophalen:1 http://archive.raspberrypi.com/debian bookworm InRelease [23,6 kB]
Ophalen:2 http://download.opensuse.org/repositories/home:/rockowitz/Raspbian_12  InRelease [1.516 B]
Geraakt:3 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Ophalen:4 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [422 kB]
Ophalen:5 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [413 kB]
Fout:2 http://download.opensuse.org/repositories/home:/rockowitz/Raspbian_12  InRelease
  De volgende ondertekeningen konden niet geverifieerd worden omdat de publieke sleutel niet beschikbaar is: NO_PUBKEY E745993F598252FB
Pakketlijsten worden ingelezen...
W: GPG-fout: http://download.opensuse.org/repositories/home:/rockowitz/Raspbian_12  InRelease: De volgende ondertekeningen konden niet geverifieerd worden omdat de publieke sleutel niet beschikbaar is: NO_PUBKEY E745993F598252FB
E: De pakketbron 'http://download.opensuse.org/repositories/home:/rockowitz/Raspbian_12  InRelease' is niet ondertekend.
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: De sleutel wordt opgeslagen in de uitgefaseerde sleutelbos (/etc/apt/trusted.gpg) in trusted.gpg, raadpleeg de afdeling UITFASERING (DEPRECATION) in apt-key(8) voor details.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: bookworm
 - Hardware: raspberry 5
 - Console Version: 23.3.1

**Additional context**
Add any other context about the problem here.

peted-davis, 2024-05-15:
Have you added custom opensuse repositories to apt/sources.list file? Your Pi is refusing to update the opensuse repositories because their GPG key has expired. That needs to be fixed before the PiConsole will update, and it is an operating system error rather than a PiConsole issue

pjmpessers, 2024-05-19:
Thanks. I indeed see that. Also not possible to update sudo apt-get update. I did clean install debian bookworm and the problem was gone. Thanks for the help.

closed 2024-05-19 by pjmpessers