peted-davis/WeatherFlow_PiConsole #146: Wont start after update...

BoatGod, 2024-06-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
boatgod@boatgod-305V4A-305V5A-3415VA:~/wfpiconsole$ wfpiconsole start
Traceback (most recent call last):
  File "/home/boatgod/wfpiconsole/main.py", line 37, in <module>
    from lib     import config as configFile
  File "/home/boatgod/wfpiconsole/lib/config.py", line 19, in <module>
    from kivy.logger    import Logger
ModuleNotFoundError: No module named 'kivy'

peted-davis, 2024-06-12:
This a strange. Can you take a backup of your `wfpiconsole.ini` file and then delete the entire `wfpiconsole` directory. Once you have done that, try installing the console again using 

`curl -sSL https://peted-davis.github.io/wfpiconsole | bash`

Does it run after the install has finished?

BoatGod, 2024-06-17:
After many hours of messing around, I came to the conclusion that wfpiconsole doesn't like the update that Kunbutu made. So, I reverted to an earlier flavor of Kunbutu and everything works again. Thanks for your work on this!

peted-davis, 2024-06-17:
Great! Glad you are up and running again. I have never tried to run the PiConsole on Kunbutu, so it is hard to comment on what the reason is. Are you running on a Raspberry Pi or a PC? If I get a chance, I will try and replicate the error

closed 2024-06-17 by peted-davis

BoatGod, 2024-06-17:
> <https://github.com/peted-davis/WeatherFlow_PiConsole/issues/146> as not
> <peted-davis/WeatherFlow_PiConsole/issue/146/issue_event/13184656143@
> github.com>

running on an old laptop, lol!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>