peted-davis/WeatherFlow_PiConsole #147: Installed and when  I run console program goes back to user prompt.,

hamgeek73, 2024-06-16:
attached is the log when I go to start.  PS:  This is my first raspberry pi so could be  rookie mistake on my part.   
[wfpiconsole.log](https://github.com/user-attachments/files/15859830/wfpiconsole.log)

**Describe the bug**
Terminal goes back to prompt.

**To Reproduce**
start app.'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: [e.g. Raspberry Pi  32 bitOS Bullseye]
 - Hardware:  Rapberry Pi 4 1GIG
 - Console Version: [just downloaded]

**Additional context**
Add any other context about the problem here.

peted-davis, 2024-06-17:
Thanks for this bug report. I've traced the issue to installing the demo configuration file during the initial setup. I should be able to get a fix in place pretty quickly. In the meantime I would respond `n` to using the demo configuration file and then you'll be taken through the full setup process without issue

peted-davis, 2024-06-17:
OK, this should now be fixed in v24.6.2. If you run the install command again, you should get a copy of the new code with the bug fix

closed 2024-06-17 by peted-davis