peted-davis/WeatherFlow_PiConsole #148: Configuration

hamgeek73, 2024-06-19:
Thanks for last fix, now having configuration issues..   Not getting the prompt to enter the tempest  or weather underground keys I entered them manually in the wpficonsole.ini along with serial number and station id.   Attached are the errors I'm getting.   Error reads like it is the station ID should be an integer and it is.   

[wfpiconsole.log](https://github.com/user-attachments/files/15894689/wfpiconsole.log)

peted-davis, 2024-06-19:
Make sure to delete any existing `wfpiconsole.ini` file. Then start the console with `wfpiconsole start`. You will be asked whether you want to install a minimal configuration file for demonstration purposes. Answer `n` to this question. Then you will be taken through the setup and asked to enter your API keys and station details. 

You will need your WeatherFlow Key and a CheckWX key (not a Weather Underground key). Please read the GitHub `README` which points you in the direction of the information you need.

hamgeek73, 2024-06-23:
Thank you  That was the trick.

 

Ed Wiest

 ***@***.***> ***@***.***

614-284-1101

closed 2024-06-23 by peted-davis