peted-davis/WeatherFlow_PiConsole #150: Install on Ubuntu 24.04.1 LTS

bleeblak8, 2024-10-24:
My WeatherFlow Console Ubuntu x86 device updated to the latest version of Ubunut LTS.  Weatherflow fails to start.
In troubleshooting, I spun up 2 brand new VMs that are using the following versions.

VM1 - Ubuntu 22.04.5 LTS
Python 3.10-venv
WeatherFlowPiConsole installs just fine.

VM2 - Ubuntu 24.04.1 LTS
Python 3.12-venv
WeatherFlowPiConsole install script fails with error - Filed building wheel for kivy.

Not sure if the kivy install script doesn't work with the version of Ubuntu and Python. 

I can certainly roll back but wanted to make you aware that it breaks in 24.04.1 LTS.

peted-davis, 2024-10-24:
Thanks for the report. I suspect there are no pre-compiled Kivy wheels for Python 3.12, and the installer is failing as it can't compile Kivy from source (missing dependencies). I suggest rolling back to 22.04 and waiting for the Kivy team to roll out updated wheels

bleeblak8, 2024-10-24:
Thanks, I ended up doing just that.  Much appreciated.

closed 2024-12-17 by peted-davis