peted-davis/WeatherFlow_PiConsole #152: PiConsole KeyError ‘clouds’

JourMic, 2025-02-14:
**Describe the bug**

Since this morning PiConsole is no longer working. When starting, I receive the error message KeyError ‘clouds’

**To Reproduce**
Steps to reproduce the behavior:

Start the app.

**Screenshots**
 
[image]

**Hardware/OS/Version (please complete the following information):**

RPI 4
Desbian 11.11

**Additional context**
Add any other context about the problem here.

peted-davis, 2025-02-15:
Thanks for the report. This should now be fixed in the latest version (v25.2.1). 

Run `wfpiconsole update` to get the latest release

closed 2025-02-15 by peted-davis

JourMic, 2025-02-15:
it works, thanks