peted-davis/WeatherFlow_PiConsole #155: Rain dashes piconsole

luistorreon, 2025-06-23:
*Describe the bug**
Hi everyone. Since last week, the PicoConsole rain chart has only been showing dashes. I've already restarted the screen, the hub, and the device, but nothing. The app is showing normal values. Any help?
Thanks.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware/OS/Version (please complete the following information):**
 - OS: Raspberry Pi OS Bullseye
 - Hardware: Pi3
 - 

[image]

peted-davis, 2025-06-23:
Yes, it seems like an API failure is currently returning `None` for the rainfall fields. Once this bug is fixed by WeatherFlow, the PiConsole should return to normal

luistorreon, 2025-06-24:
I hope it gets fixed soon. The problem still persists 😔

luistorreon, 2025-06-27:
The problem has already been fixed. I just reset the display. Thanks, Tempest.

[image]

closed 2025-06-27 by luistorreon